gopcua/opcua #593: Refused to open SecureChannel, when using x509 self signed certificate

Sandwich-Theorem, 2022-08-02:
hello, 
I am working on connectiong to an opcua server via a x509 certificate, which is created for this purpose. 
When running the programm I get the following debug statements:

```
debug: uacp: connecting to opc.tcp://192.168.0.90:48020
debug: uacp 2: start HEL/ACK handshake
debug: uacp 2: sent HELF with 74 bytes
debug: uacp 2: recv ACKF with 28 bytes
debug: uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 1816 bytes
debug: uacp 2: recv ERRF with 16 bytes
debug: uasc 2: readChunk EOF
debug: uasc 2: failed to open a new secure channel
debug: uacp 2: close
panic: EOF
failed to connect to server
```

To initialize the client: 
```
func NewClient_securityModePolicy_generateCertPrivKey(endpoint string, mode string, policy string, applicationURI string,
            company string, country string, locality string, streetAddress string, postalCode string) (*opcua.Client, context.Context){

    // check policy 
    if policy != "Basic128Rsa15" && policy != "Basic256" && policy != "Basic256Sha256" {
        panic("Security is not valid");
    }

    secpolicy := ua.SecurityPolicyURIPrefix + policy;

    var secmode ua.MessageSecurityMode;

    // check security mode
    switch mode {
        case "Sign", "sign":
            secmode = ua.MessageSecurityModeSign;

        case "SignAndEncrypt", "signandencrypt":
            secmode = ua.MessageSecurityModeSignAndEncrypt;

        default:
            panic("Security mode is not valid. must be Sign or SignAndEncrypt\n");

    }
    ctx := context.Background();

    //get possible endpoints to connect to
    endpoints, err := opcua.GetEndpoints(ctx, endpoint);
    if err != nil{
        panic(err);
    }
    // choose endpoint, which requires selected security parameters
    ep := opcua.SelectEndpoint(endpoints, secpolicy, secmode);
    if ep == nil {
        panic("Failed to find suitable endpoint");
    }

    // create self signed x509 certificate 
    x509_filename, priv_key_filename := create_x509cert_privateKey(company, country, locality,
                                            streetAddress, postalCode, applicationURI);
    
    // load the x509 certificate and the corresponding private key
    c, err := tls.LoadX509KeyPair(x509_filename, priv_key_filename);
    if err != nil {
        panic(err.Error() + "\nfailed to load certificate");
    }

    pk, ok := c.PrivateKey.(*rsa.PrivateKey);
    if !ok {
        panic("Invalid private key");
    }

    cert := c.Certificate[0];

// sum up the options for using the certificate
    opts := []opcua.Option{
        opcua.Certificate(cert),
        opcua.PrivateKey(pk),
        opcua.AuthCertificate(cert),
        opcua.SecurityFromEndpoint(ep, ua.UserTokenTypeCertificate),
    }
// make sure that the endpoint supports the choice
    err = validateEndpointConfig(endpoints,secpolicy, secmode,
        ua.UserTokenTypeCertificate);
    if err != nil{
        panic("error validating input " + err.Error());
    }

    client := opcua.NewClient(ep.EndpointURL, opts...);

    connect(client, ctx);

    return client, ctx;
}
```

To create the x509 certificate: 
```
func create_x509cert_privateKey(company string, country string, locality string,
                                streetAddress string, postalCode string, applicationURI string) (string, string){

    serialNumberLimit := new(big.Int).Lsh(big.NewInt(1), 128)
    serialNumber, err := rand.Int(rand.Reader, serialNumberLimit)
    if err != nil {
        panic("could not generate serial number");
    }

    ca := &x509.Certificate{
        // use SHA256 for the case SignAndEncrypt
        SignatureAlgorithm: x509.SHA256WithRSAPSS,
        SerialNumber: serialNumber,
        Subject: pkix.Name{
            Organization: []string{company},
            Country:      []string{country},
            Province:     []string{""},
            Locality:     []string{locality},
            StreetAddress:[]string{streetAddress},
            PostalCode:   []string{postalCode},
        },
        IPAddresses:  []net.IP{net.IPv4(127, 0, 0, 1), net.IPv6loopback, net.IPv4(192, 168, 0, 90)},
        NotBefore:  time.Now(),
// 4 months
        NotAfter:   time.Now().AddDate(0, 4, 0),
        ExtKeyUsage:[]x509.ExtKeyUsage{x509.ExtKeyUsageClientAuth, x509.ExtKeyUsageServerAuth},
        KeyUsage:   x509.KeyUsageContentCommitment | x509.KeyUsageKeyEncipherment | x509.KeyUsageDigitalSignature | x509.KeyUsageDataEncipherment | x509.KeyUsageCertSign,
        BasicConstraintsValid: true,
    }

    hosts := strings.Split(applicationURI, ",")
        for _, h := range hosts {
            if ip := net.ParseIP(h); ip != nil {
                (*ca).IPAddresses = append((*ca).IPAddresses, ip)
            } else {
                (*ca).DNSNames = append((*ca).DNSNames, h)
            }
            if uri, err := url.Parse(h); err == nil {
                (*ca).URIs = append((*ca).URIs, uri)
            }
        }
// create public and private key via rsa
    caPrivKey, err := rsa.GenerateKey(rand.Reader, 2048);
    if err != nil {
        panic(err);
    }
// self sign the certificate
    derBytes, err := x509.CreateCertificate(rand.Reader, ca, ca, &caPrivKey.PublicKey, caPrivKey)
    if err != nil {
        panic(err);
    }

// save the certificate to a file
    certOut, err := os.Create("cert.pem");
    if err != nil {
        panic("Failed to create cert.pem file" + err.Error());
    }
// pem encode the file
    if err := pem.Encode(certOut, &pem.Block{
        Type: "CERTIFICATE",
        Bytes: derBytes,
    }); err != nil {
        panic("Failed to write data to cert.pem file "+err.Error());
    }

    if err := certOut.Close(); err != nil {
        panic("error closing cert.pem file "+ err.Error());
    }
// save the private key to a file
    keyOut, err := os.OpenFile("key.pem", os.O_WRONLY | os.O_CREATE | os.O_TRUNC, 0600);
    if err != nil {
        panic(err.Error() + "\n failed to create key.pem file");
    }
// pem encode the file
    if err := pem.Encode(keyOut, &pem.Block{
        Type: "RSA PRIVATE KEY",
        Bytes: x509.MarshalPKCS1PrivateKey(caPrivKey),
    }); err != nil {
        panic(err.Error() + "\n failed to write data to key.pem");
    }

    if err:= keyOut.Close(); err != nil {
        panic(err.Error() + "\n error closing key.pem file");
    }

    return "cert.pem", "key.pem";
}
```

The validateInput function: 
```
func validateEndpointConfig(endpoints []*ua.EndpointDescription, secPolicy string, secMode ua.MessageSecurityMode, authMode ua.UserTokenType) error {
    for _, e := range endpoints {
        if e.SecurityMode == secMode && e.SecurityPolicyURI == secPolicy {
            for _, t := range e.UserIdentityTokens {
                if t.TokenType == authMode {
                    return nil
                }
            }
        }
    }

    err := errors.Errorf("server does not support an endpoint with security : %s , %s", secPolicy, secMode)
    return err
}
```

Can anyone please tell me where my mistake lies. 
Thank you very much.

kung-foo, 2022-08-04:
Hi @Sandwich-Theorem, from your code it looks like you are generating a new certificate each time the program runs. Most OPC-UA servers do not automatically trust unknown certificates. So usually connecting via x509 requires several steps:

1) attempt a connection from the client, which will fail
2) go to the OPC-UA server and "accept" or trust the unknown certificate
3) attempt a connection from the client, which should succeed

jackchenjc, 2023-08-15:
Hi @kung-foo,
I have similar problem, but I did use the same certificates which was trusted by the server to attempt a connection from the client.

Not sure what I missed and I have just a silly question here.
As far as I know, `gopcua/opcua` supports these three functions of type Option:
```
func Certificate(cert []byte) Option
func PrivateKey(key *rsa.PrivateKey) Option
func AuthCertificate(cert []byte) Option
```
If we use different cert for `client` and `user auth`, which private key should we pass to `opcua.PrivateKey()`?

Thanks.

closed 2025-04-23 by github-actions[bot]