gopcua/opcua #604: [HELP] Import information model to server

xujiesh0510, 2022-09-24:
Hi. is there any way to import  information model  (a nodeset2.xml file )  to server    ?

opcua c# server and   opcua  c  server  can do this.

 nodeset2.xml  looks like this:

```

<?xml version="1.0" encoding="utf-8"?>
<UANodeSet xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xmlns:xsd="http://www.w3.org/2001/XMLSchema" xmlns="http://opcfoundation.org/UA/2011/03/UANodeSet.xsd">
  <NamespaceUris>
    <Uri>http://opcfoundation.org/BatchPlant</Uri>
  </NamespaceUris>
  <Models>
    <Model ModelUri="http://opcfoundation.org/BatchPlant">
      <RequiredModel ModelUri="http://opcfoundation.org/UA/" XmlSchemaUri="http://opcfoundation.org/UA/2008/02/Types.xsd" />
    </Model>
  </Models>
  <Aliases>
    <Alias Alias="Boolean">i=1</Alias>
   .....
 
 
  </Aliases>
  <UAObject NodeId="ns=1;i=1" BrowseName="1:工厂" SymbolicName="allfactory">
    <DisplayName>工厂</DisplayName>
    <Description>Contains all instances of animals</Description>
    <References>
      <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
    </References>
  </UAObject>
  <UAObjectType NodeId="ns=1;i=2" BrowseName="1:CustomEvent">
    <DisplayName>CustomEvent</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=13</Reference>
      <Reference ReferenceType="HasSubtype" IsForward="false">i=2133</Reference>
    </References>
  </UAObjectType>
```

thanks

closed 2022-09-24 by xujiesh0510

piperoc, 2024-01-29:
@xujiesh0510  you flagged this as completed, but there is no answer to your original questions. Did you resolve this? It doesn't look like there is a NodeSet import facility... it would be very nice. Thanks for expanding on this.

danomagnum, 2024-01-30:
The newest commits of the [server branch ](https://github.com/gopcua/opcua/tree/server) has an ImportNodeSet function on the server object.

You can see an example of it in the base server initialization logic where it imports the base opc ua nodeset from an embedded nodeset2.xml file.

https://github.com/gopcua/opcua/blob/c449b7ce33904016dd76ec6f2a0aa798cded7ee4/server/server.go#L145

I haven't tested it with other nodeset files but if you give it a try, let me know if it works or not.

xujiesh0510, 2024-01-31:
> @xujiesh0510 you flagged this as completed, but there is no answer to your original questions. Did you resolve this? It doesn't look like there is a NodeSet import facility... it would be very nice. Thanks for expanding on this.

finally，I had to  use open62541 ,which is a C library. other languages' implementaion is not fully supported.