gopcua/opcua #605: Reading data by Path

pedro-ricardo, 2022-09-28:
Hello there, 

I'm currently able to use the library to read data from a nodeID like `ns=1;i=8011`, but I wish to pass the path like `ns=1;/Objects/My_Variable`.

If I understood correctly the lib can Browse the server but it can not `TranslateBrowsePathsToNodeIds`. 

I'm interested in implementing this, could you guys point me a starting point?

zskvd, 2022-10-06:
You should translate the browse path into nodeId first, this can be done by using the TranslateBrowsePathInNamespaceToNodeID API.  examples/translate may offer some help.

pedro-ricardo, 2022-10-06:
Thank you very much!
We've added this issue to the list of enhancements in our project.

Some team member should tackle this eventually.
Hopefully soon :smile:

closed 2025-04-23 by github-actions[bot]