gopcua/opcua #606: Semicolon in node name

ivanterekh, 2022-09-29:
OPC UA standard does not set any limits on Name of string id node. Therefore the nodes containing semicolon should be allowed. For example: `"ns=1;s=Folder;Object;Variable"` .

The current implementation explicitly disallows nodes with more than 1 semicolon:

https://github.com/gopcua/opcua/blob/892f58798e2c9bd0de67f6c5c3b410602f928457/ua/expanded_node_id.go#L144-L152

I see no reason to limit node names in client if an eventual server might allow them. [OPC PLC server](https://github.com/Azure-Samples/iot-edge-opc-plc) is an example of server that actually has a node with an additional semicolon in the node identifier.

pedro-ricardo, 2022-09-29:
If it is a valid OPCUA name, than a escape character seems to be needed. This seems like a nice addition :+1:

kung-foo, 2022-09-29:
yep. can confirm that Azure's server includes this beauty:
```
nsu=http://microsoft.com/Opc/OpcPlc/;s=Special_"!§$%&/()=?`´\\+~*\'#_-:.;,<>|@^°€µ{[]}
```

PR #607 (2022-09-29): relax the node id parser a little bit
fixes #606
closed 2022-09-29 by kung-foo