gopcua/opcua #609: subscription alwarys fail after some time

zskvd, 2022-10-06:
I tried to subscribe some items on the Prosys Simulation Server, using the examples/subsribe.go, with option as follows:
go run examples/subscribe/subscribe.go -endpoint opc.tcp://uademo.prosysopc.com:53530/OPCUA/SimulationServer -node "ns=3;i=1001" -mode None -debug True

magiconair, 2022-10-06:
Does the value change? Sounds like session timeout. Try subscribing to `i=2258` (current time)

zskvd, 2022-10-07:
> Does the value change? Sounds like session timeout. Try subscribing to `i=2258` (current time)

Yes, I also did this, leads to the same error,here is the debugging messages:

```
debug: uasc 2/62: send *ua.PublishRequest with 106 bytes
MonitoredItem with client handle 42 = 2022-10-07 04:18:48.395 +0000 UTC
debug: uacp 2: recv MSGF with 140 bytes
debug: uasc 2/62: recv MSGF with 140 bytes
debug: uasc 2/62: recv *ua.PublishResponse
debug: uasc 2/62: sending *ua.PublishResponse to handler
debug: publish: PublishResponse: {"ResponseHeader":{"Timestamp":"2022-10-07T04:18:49.396Z","RequestHandle":62,"ServiceResult":0,"ServiceDiagnostics":{"EncodingMask":0,"SymbolicID":0,"NamespaceURI":0,"Locale":0,"LocalizedText":0,"AdditionalInfo":"","InnerStatusCode":0,"InnerDiagnosticInfo":null},"StringTable":null,"AdditionalHeader":{"EncodingMask":0,"TypeID":{"NodeID":"i=0","NamespaceURI":"","ServerIndex":0},"Value":null}},"SubscriptionID":1172,"AvailableSequenceNumbers":[56],"MoreNotifications":false,"NotificationMessage":{"SequenceNumber":56,"PublishTime":"2022-10-07T04:18:49.396Z","NotificationData":[{"EncodingMask":1,"TypeID":{"NodeID":"i=811","NamespaceURI":"","ServerIndex":0},"Value":{"MonitoredItems":[{"ClientHandle":42,"Value":{"EncodingMask":13,"Value":{},"Status":0,"SourceTimestamp":"2022-10-07T04:18:49.396Z","SourcePicoseconds":0,"ServerTimestamp":"2022-10-07T04:18:49.396Z","ServerPicoseconds":0}}],"DiagnosticInfos":null}}]},"Results":[0],"DiagnosticInfos":null}
debug: publish: notAcked=[]
debug: publish: notif: 56
debug: publish: pendingAcks=[{"SubscriptionID":1172,"SequenceNumber":56}]
debug: publish: PublishRequest: {"RequestHeader":null,"SubscriptionAcknowledgements":[{"SubscriptionID":1172,"**SequenceNumber":56**}]}
MonitoredItem with client handle 42 = 2022-10-07 04:18:49.396 +0000 UTC
debug: uasc 2/63: send *ua.PublishRequest with 106 bytes
debug: uasc 2: Close()
debug: client: monitor: done
debug: publish: PublishResponse: null
debug: publish: error: unexpected error. Do we need to stop the publish loop?: context canceled
debug: sub: error: context canceled
debug: sub: pause
debug: sub: pause: ctx.Done()
debug: sub: done
debug: uasc 2/66: send *ua.CloseSecureChannelRequest with 57 bytes
debug: uacp 2: close
debug: uasc 2: readChunk EOF
```

As you see, the last successful response sequence is 56, and then fail.

magiconair, 2022-10-07:
I think it's because of the way the example is written. There is a 60 sec timeout in the code that isn't very obvious. I think I stumbled over this as well. 

https://github.com/gopcua/opcua/blob/main/examples/subscribe/subscribe.go#L35-L39

zskvd, 2022-10-07:
Sorry, I'm so stupid that I missed the import code of the example;
```

	// add an arbitrary timeout to demonstrate how to stop a subscription
	// with a context.
	d := 60 * time.Second
	ctx, cancel := context.WithTimeout(context.Background(), d)
	defer cancel()
```

magiconair, 2022-10-07:
Yeah, it's not obvious if you don't specifically look for it. I think none of the other examples are like this. We should probably log this to make this more obvious.

zskvd, 2022-10-07:
> 

Lol, 😂.   I found the problem at the time when you figure out the real problem!

magiconair, 2022-10-07:
I'll add a log line. :)

zskvd, 2022-10-07:
> I think it's because of the way the example is written. There is a 60 sec timeout in the code that isn't very obvious. I think I stumbled over this as well.
> 
> https://github.com/gopcua/opcua/blob/main/examples/subscribe/subscribe.go#L35-L39

Lol, 😂.   I found the problem at the time when you figure out the real problem!

magiconair, 2022-10-07:
https://github.com/gopcua/opcua/commit/3618f0ec0fcc2f33ad91303374404bb6daa8e44a

magiconair, 2022-10-07:
Can you have a look whether this makes it obvious enough? It's on `main`

zskvd, 2022-10-07:
> Can you have a look whether this makes it obvious enough? It's on `main`
[image]
I think it's not very obvious, if the server is fast enough, all the messages flashed through the screen . May be you can give  some pause time after this log message appears.😄

closed 2022-10-07 by zskvd