gopcua/opcua #613: Why does it always return EOF when receiving an ERR type of return packet？

zmj159809, 2022-10-27:
Obviously the server did not disconnect the connection, but just returned an error.
I can't even tell from the log what type of error it is
Is it necessary to judge the length equal to 0 in readChunk?
```
	// read a full message from the underlying conn.
	b, err := s.c.Receive()
	if err == io.EOF || len(b) == 0 {
		return nil, io.EOF
	}
```

```
debug: uacp: connecting to opc.tcp://BA-SERVER1
debug: uacp 3: start HEL/ACK handshake
debug: uacp 3: sent HELF with 52 bytes
debug: uacp 3: recv ACKF with 28 bytes
debug: uacp 3: server has no chunk limit. Using 512
debug: uacp 3: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x1000000, MaxChunkCount:0x200}
debug: uasc 3/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug: uacp 3: recv OPNF with 135 bytes
debug: uasc 3/1: recv OPNF with 135 bytes
debug: uasc 3/1: recv *ua.OpenSecureChannelResponse
debug: uasc 3/1: sending *ua.OpenSecureChannelResponse to handler
debug: uasc 3: received security token. channelID=14 tokenID=1 createdAt=2022-10-27T03:44:34Z lifetime=1h0m0s
debug: uasc 3: security token expires at 2022-10-27T04:59:34Z. channelID=14 tokenID=1
debug: uasc 3: security token is refreshed at 2022-10-27T04:28:05Z (45m0s). channelID=14 tokenID=1
debug: uasc 3/2: send *ua.CreateSessionRequest with 259 bytes
debug: uacp 3: recv ERRF with 16 bytes
debug: uasc 3: readChunk EOF
debug: uasc 3: Close()
debug: uacp 3: close
```

This is the information in wireshark
[image]

Klikini, 2025-04-03:
I'm also experiencing this.

Wireshark shows `BadSecurityChecksFailed`:

[image]

But the debug log shows nothing useful, and the error returned by `Client.Connect` is just `EOF`:

```
debug: uacp: connecting to opc.tcp://EDGE-340595:4840/HEIDENHAIN/NC
debug: uacp 2: start HEL/ACK handshake
debug: uacp 2: sent HELF with 72 bytes
debug: uacp 2: recv ACKF with 28 bytes
debug: uacp 2: server has no chunk limit. Using 512
debug: uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x1000000, MaxChunkCount:0x200}
debug: uasc 2/1: send *ua.OpenSecureChannelRequest with 1539 bytes
debug: uacp 2: recv ERRF with 16 bytes
debug: uasc 2: readChunk EOF
debug: uasc 2: failed to open a new secure channel
debug: uacp 2: close
2025/04/03 16:53:19 error connecting to OPC-UA server: EOF
```

magiconair, 2025-04-07:
Looks like the `uacp` error frame isn't bubbled up.