gopcua/opcua #616: log.Fatal called when a certificate fails to load

jrtitus, 2022-11-10:
This causes any process using gopcua/opcua to exit without warning and is caused by either the referenced file not existing, or a decoding failure. 

Can it be handled more cleanly like logging an error and returning? Or perhaps returning `(Option, error)` from `CertificateFile`?

https://github.com/gopcua/opcua/blob/3618f0ec0fcc2f33ad91303374404bb6daa8e44a/config.go#L268-L271

It looks like the same thing happens in the `RemoteCertificateFile`, `PrivateKeyFile`, and `setCertificate` functions as well. I also see TODO comments in `AuthAnonymous` and `AuthUsername` questioning the use of `Fatal`. My suggestion would be no.

magiconair, 2022-11-11:
Eh, `log.Fatal` in a library is never OK. Sorry about that. Will fix.

```sh
$ ack -l log.Fatal | grep -v examples
cmd/status/main.go
cmd/id/main.go
cmd/service/main.go
config.go
```

magiconair, 2022-11-11:
Ah, I remember why I did this 3y ago. No good option to bubble up errors for config options. I'll rework the config option.

jrtitus, 2022-11-11:
> Ah, I remember why I did this 3y ago. No good option to bubble up errors for config options. I'll rework the config option.

Awesome. Thank you very much for following up so quickly on this.

magiconair, 2022-11-14:
So we have two options to solve this:

1. We bubble the error up during `Dial` and/or `Connect` which does not require us to change any API
2. We change `NewClient` to return an error which breaks all existing code. Cleaner but more disruptive.
3. We do 1. until v0.5.0 where we have to break the API anyway and then update `NewClient`.

I'm for option 3. @kung-foo any opinion?

closed 2022-11-17 by magiconair

jrtitus, 2022-12-07:
Hi @magiconair. Just curious - when do you plan to make the `v0.3.8` release?

jrtitus, 2022-12-08:
> sorry, dropped the ball here.

No worries!

> done

Thanks! :)

PR #618 (2022-11-17): config: bubble up errors from configuration
The current config code has some options which can have errors. The current code calls `log.Fatal` which terminates the application instead of returning the error.

The correct solution would be to return the error in ApplyConfig and then return it in NewClient. However, this would break all existing clients.

We therefore postpone this breaking change to v0.5.0 when we update the API with breaking changes anyway and in the meantime bubble the error up in the Connect and Dial method.

Fixes #616
commit bc8fc63c (2022-11-17): config: bubble up errors from configuration

The current config code has some options which can have errors. The
current code calls `log.Fatal` which terminates the application instead
of returning the error.

The correct solution would be to return the error in ApplyConfig and
then return it in NewClient. However, this would break all existing
clients.

We therefore postpone this breaking change to v0.5.0 when we update the
API with breaking changes anyway and in the meantime bubble the error up
in the Connect and Dial method.

Fixes #616