gopcua/opcua #620: Client.State() seems wrong

sibebleuze, 2022-11-16:
# Steps to reproduce:
1. Connect to some OPC-UA server
  - I am using opc.tcp://historian.factry.dev:54845 as endpoint, but I'm not sure if it is available outside our network.
2. Find a node where calling `Node.Attributes(ua.AttributeIDArrayDimensions)` returns a `StatusBadEncodingLimitsExceeded` (not important right now)
  - For the server above, I encountered this at node 'ns=1;s=Sine_1'
3. At this point, `Client.State() == opcua.Connected` evaluates to `true`
4. Making another call with the client, for example repeating `Node.Attributes(ua.AttributeIDArrayDimensions)`, returns `StatusBadServerNotConnected`

# Expected behaviour
I want to try to avoid the StatusBadServerNotConnected by catching `Client.State() != opcua.Connected` and finding a way to reconnect at that point, but as the State seems to not reflect the actual state of the client, I cannot do this.

andreapicaro, 2023-03-07:
Hello, we are experiencing exactly the same issue. Reading attribute ua.AttributeIDArrayDimensions produce error StatusBadServerNotConnected and after that channel is closed and any other operation fails. In our case we'd like to solve the reading issue since we need to read that attribute. Did you manage to find a solution?
Thanks

sibebleuze, 2024-09-08:
I did not, unfortunately.  
At the time, I was only experimenting with both golang and opcua and having a look at what is all available in the opcua attributes. I didn't really need the AttributeIDArrayDimensions so I think I just avoided it during my further tests.

sibebleuze, 2025-04-13:
Closing as I am unable to reproduce the issue with the latest version, both the StatusBadEncodingLimitsExceeded and by extension the resulting seemingly inconsistent client.State().

closed 2025-04-13 by sibebleuze