gopcua/opcua #625: OpenSecureChannelRequest return readChunk EOF

whutwxn, 2022-12-20:
I connect to INOVANCE PLC,
go to step:uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
It return : debug: uasc 1: readChunk EOF
I compared the request packets sent by gopcua and Ua Expert application using wireshark, and the screenshot is as follows:
gopcua:
[image]
Ua Expert:
[image]

the Ua Expert send 133bytes with the clientnonce=0x00,but gopcua send 132bytes without clientonce.
Is it the gopcua`s bug or is it the PLC`s bug?

whutwxn, 2022-12-20:
when i add some code to the secure_channel.go line 533:
	//if len(localNonce) == 0 {
	//	var bt []byte
	//	bt = append(bt, 0x00)
	//	localNonce = bt
	//}
it works!

kung-foo, 2022-12-27:
similar issue: https://github.com/FreeOpcUa/python-opcua/issues/393

kung-foo, 2022-12-27:
My initial read of this is that this is a bug in the OPC-UA server.

kung-foo, 2022-12-27:
Can you try the following patch:

```patch
diff --git a/uapolicy/securitypolicy.go b/uapolicy/securitypolicy.go
index 3b30933..d211aa9 100644
--- a/uapolicy/securitypolicy.go
+++ b/uapolicy/securitypolicy.go
@@ -148,7 +148,11 @@ func (e *EncryptionAlgorithm) NonceLength() int {
 }
 
 func (e *EncryptionAlgorithm) MakeNonce() ([]byte, error) {
-       b := make([]byte, e.NonceLength())
+       l := e.NonceLength()
+       if l == 0 {
+               return nil, nil
+       }
+       b := make([]byte, l)
        // note: we use `rand.Reader` instead of `rand.Read(...)` to ensure that we don't accidentally switch to using
        // math/rand (which has a default, fixed seed). Only crypto/rand exposes a global `io.Reader` var.
        _, err := io.ReadFull(rand.Reader, b)
```

This will send a `null` bytestring nonce versus an empty (zero-length) bytestring.

whutwxn, 2023-01-09:
ok,thanks

closed 2023-01-09 by whutwxn

magiconair, 2023-01-09:
@whutwxn Does that mean the fix worked?