gopcua/opcua #635: AttributesWithContext(ua.AttributeIDAccessLevel) always returns None, should be CurrentRead/CurrentWrite

JulianKrmr, 2023-02-02:
Following the browse example provided in the examples folder, the value of AttributeIDAccessLevel is always 0, I tested it with a Simulation Server with the permission set to CurrentRead and CurrentWrite and I can use both write and read on the node.

JulianKrmr, 2023-03-01:
Can I get a reply to this please? :)

magiconair, 2023-03-01:
I'm currently on vacation but can have a look next week when I'm back. @kung-foo maybe you have some time before that?

JulianKrmr, 2023-03-22:
Push
I would be very thankful for help on this one!

magiconair, 2023-03-22:
I'll try to have a look tomorrow

magiconair, 2023-03-23:
Hmm, can't reproduce this...

[image]

[image]

[image]

magiconair, 2023-03-23:
Wait. I'm testing the wrong method

magiconair, 2023-03-23:
Hmm, still works ...

```go
// Copyright 2018-2020 opcua authors. All rights reserved.
// Use of this source code is governed by a MIT-style license that can be
// found in the LICENSE file.

package main

import (
	"context"
	"flag"
	"log"

	"github.com/gopcua/opcua"
	"github.com/gopcua/opcua/debug"
	"github.com/gopcua/opcua/ua"
)

func main() {
	var (
		endpoint = flag.String("endpoint", "opc.tcp://localhost:4840", "OPC UA Endpoint URL")
		nodeID   = flag.String("node", "", "NodeID to read")
	)
	flag.BoolVar(&debug.Enable, "debug", false, "enable debug logging")
	flag.Parse()
	log.SetFlags(0)

	ctx := context.Background()

	c := opcua.NewClient(*endpoint, opcua.SecurityMode(ua.MessageSecurityModeNone))
	if err := c.Connect(ctx); err != nil {
		log.Fatal(err)
	}
	defer c.CloseWithContext(ctx)

	id, err := ua.ParseNodeID(*nodeID)
	if err != nil {
		log.Fatalf("invalid node id: %v", err)
	}

	req := &ua.ReadRequest{
		MaxAge: 2000,
		NodesToRead: []*ua.ReadValueID{
			{NodeID: id, AttributeID: ua.AttributeIDAccessLevel},
		},
		TimestampsToReturn: ua.TimestampsToReturnBoth,
	}

	resp, err := c.ReadWithContext(ctx, req)
	if err != nil {
		log.Fatalf("Read failed: %s", err)
	}
	if resp.Results[0].Status != ua.StatusOK {
		log.Fatalf("Status not OK: %v", resp.Results[0].Status)
	}
	log.Printf("%#v", resp.Results[0].Value.Value())

	resp2, err := c.Node(id).AttributesWithContext(ctx, ua.AttributeIDAccessLevel)
	if err != nil {
		log.Fatalf("AttributesWithContext failed: %s", err)
	}
	if resp2[0].Status != ua.StatusOK {
		log.Fatalf("Status not OK: %v", resp2[0].Status)
	}
	log.Printf("%#v", resp2[0].Value.Value())
}
```

[image]

JulianKrmr, 2023-03-23:
Thanks alot for checking, I guess there must be a mistake by my side then. I will look it up on my side again and then will give an update here.

magiconair, 2023-03-23:
No worries. I'll close it. Please re-open if necessary.