gopcua/opcua #636: OPC-UA server can have multiple certificates

geoynomous, 2023-02-08:
The OPC-UA might announce multiple certificates - concatenated. Then

``x509.ParseCertificate(cert)`` does not work - it will report a ``x509: trailing data`` error. 

It is better to use ``x509.ParseCertificates(certsByteArray)`` (**Plural!**) and then to chose one of them. The first? 

Honestly I do not understand why some servers provide multiple certs - maybe there should be a way to select one of them? You know it better ...

geoynomous, 2023-02-27:
Any comment here?

geoynomous, 2023-10-30:
This is still open

allenlu1990, 2023-12-27:
and why Private Key must be RSA PRIVATE KEY?

geoynomous, 2023-12-27:
@allenlu1990 I understand your question - but this is unrelated to this issue. It might be a requirement by the standard. Here we are dealing with the fact that multiple certificates are announced.

closed 2025-04-22 by github-actions[bot]