gopcua/opcua #650: How can i get every nodeid when use batch list to read?

CHN-STUDENT, 2023-05-10:
Hi, now i have i question, i try to read opc data by using batchlist, but when i try to parse datas, i do not know which data from original nodeid, so could you tell me how can i get its original nodeid?

```
if len(resp.Results) > 0 {
  for _, item := range resp.Results {
    // how can i get every item's original nodeid?
  }
}
```

kung-foo, 2023-05-24:
Depends on the OPC-UA server. I usually just chunk the node list before sending read requests.

Regarding subscriptions, check out:

https://github.com/gopcua/opcua/blob/main/examples/subscribe/subscribe.go
https://github.com/gopcua/opcua/blob/main/examples/monitor/monitor.go

CHN-STUDENT, 2023-06-15:
OK, i try to use page slice to deal with nodeIDs List, thanks for your help!

closed 2023-06-15 by CHN-STUDENT