gopcua/opcua #656: How do you use ua.FilterOperatorOfType correctly?

Sandwich-Theorem, 2023-05-31:
My goal is to filter my events by their type (e.g BaseType, AlarmConditionType or SystemEventType) on the server to avoid unnecessary network traffic and complexity in the client. 
To achieve this, I wrote a ContentFilterElement: 
```golang
func getTypeOfContentFilter(eventtype uint32) *ua.ContentFilterElement{
	return &ua.ContentFilterElement{
		FilterOperator: ua.FilterOperatorOfType,
		FilterOperands: []*ua.ExtensionObject{
			{
				EncodingMask: 1,
				TypeID: &ua.ExpandedNodeID{
					NodeID: ua.NewNumericNodeID(0,id.LiteralOperand_Encoding_DefaultBinary),
				},
				Value: ua.LiteralOperand{
					Value: ua.MustVariant(ua.NewNumericNodeID(0, eventtype)),
				},
			},
		},
	}
}
``` 
The parameter eventtype is the corresponding id from the id pkg (e.g. id.AlarmConditionType).
Opc ua standard 7.4.3 Table 120 states, the first operator has to resolve to a NodeID. In consequence, I tried a LiteralOperand with an internal NodeID, LiteralOperand with just the uint32 id value and SimpleAttributOperand with the TypeDefinitionID. I did make sure to use the correspoind default binary encoding.
The integration of the filter in the where clause seems correct because my other content filter, using the severity, works. 
I appreciate any help, thank you very much.

magiconair, 2023-06-15:
Maybe try it like this?

```go
func getTypeOfContentFilter(eventtype uint32) *ua.ContentFilterElement {
	return &ua.ContentFilterElement{
		FilterOperator: ua.FilterOperatorOfType,
		FilterOperands: []*ua.ExtensionObject{
			ua.NewExtensionObject(&ua.LiteralOperand{
				Value: ua.MustVariant(ua.NewNumericNodeID(0, eventtype)),
			}),
		},
	}
}
```

magiconair, 2023-06-15:
I'll close this but please re-open if necessary or start a discussion.

closed 2023-06-15 by magiconair