gopcua/opcua #658: EOF error, Basic256Sha256

almondnuggets, 2023-06-05:
I have tried Basic256Sha256 sec policy with beckhoff plc twin cat, generated the certificates using the crypto.go provided in the examples. but it doesnt go beyond that. 

Enter password: 
secMode:  MessageSecurityModeSignAndEncrypt secPolicy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
Using config:
Endpoint: opc.tcp://DESKTOP-xxxx:4840
Security mode: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256, MessageSecurityModeSignAndEncrypt
Auth mode : UserTokenTypeUserName
EOF

happening somewhere over here:
c := opcua.NewClient(*endpoint, opts...)
	if err := c.Connect(ctx); err != nil {
		log.Fatal(err)
	}
	defer c.CloseWithContext(ctx)

Does anybody else can guide me what I am missing here?

PS : the same certificates I use with python free opcua and the same server everything works normal.

magiconair, 2023-06-05:
Could this be related to https://github.com/gopcua/opcua/pull/648 ?

almondnuggets, 2023-06-05:
I created the certificates using crypto.go so the key length should be 2048, dont you think so?

magiconair, 2023-06-05:
Just checking. Can you run this with `OPC_DEBUG=on` and/or capture a tcpdump? You can find me on Keybase and share the files and the cert if you want.

MrWinkyTysilio, 2023-08-09:
Hello,

Were you able to solve your problem ? 
I think I have exactly the same error using telegraf, when I tried the Basic256Sha256 with my BECKHOFF OPCUA Server.

MrWinkyTysilio, 2023-09-06:
Hello,

Maybe I find a solution: 

The problem is the following one: telegraf is using a length of 2048 for the certificat.
Beckhoff server is using a 4096.

To fix it you have to do :

    delete certificates in: Server\PKI\CA\own\certs & Server\PKI\CA\own\private
    update Server\TcUaServerConfig.xml, change keylength to 2048
    restart UA server

yang750418, 2024-01-22:
@magiconair Hello

I use Telegraf OPC UA plugin, which uses gopcua.
I am currently encountering the same situation. Using security and OPCUA to subscribe, EOF occurs when Nodes exceeds 1000 or 1500.
the logs..
```
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Connecting OPC UA Client to server
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp: connecting to opc.tcp://192.168.1.XX:48050
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: start HEL/ACK handshake
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: sent HELF with 60 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv ACKF with 28 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv OPNF with 136 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: recv OPNF with 136 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: recv *ua.OpenSecureChannelResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: received security token. channelID=2437948123 tokenID=1 createdAt=2024-01-19T06:57:35Z lifetime=1h0m0s
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: security token is refreshed at 2024-01-19T07:42:41Z (45m0s). channelID=2437948123 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: security token expires at 2024-01-19T08:12:35Z. channelID=2437948123 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: send *ua.GetEndpointsRequest with 97 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv MSGF with 3690 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: recv MSGF with 3690 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: recv *ua.GetEndpointsResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: sending *ua.GetEndpointsResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: Close()
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: close
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: readChunk EOF
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Configuring OPC UA connection options
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Loading cert/key from /etc/telegraf/ua.pem//etc/telegraf/ua_key.pem
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] security policy from configuration 
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp: connecting to opc.tcp://192.168.1.XX:48050
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: start HEL/ACK handshake
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: sent HELF with 60 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv ACKF with 28 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: send *ua.OpenSecureChannelRequest with 1905 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv OPNF with 1901 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: setting securityPolicy to http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: recv OPNF with 1901 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: recv *ua.OpenSecureChannelResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: sending *ua.OpenSecureChannelResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: received security token. channelID=2437948124 tokenID=1 createdAt=2024-01-19T06:57:35Z lifetime=1h0m0s
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: security token is refreshed at 2024-01-19T07:42:41Z (45m0s). channelID=2437948124 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: security token expires at 2024-01-19T08:12:35Z. channelID=2437948124 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: send *ua.CreateSessionRequest with 1584 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 5408 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: recv MSGF with 5408 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: recv *ua.CreateSessionResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: sending *ua.CreateSessionResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: send *ua.ActivateSessionRequest with 464 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 144 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: recv MSGF with 144 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: recv *ua.ActivateSessionResponse
[... cut ...]
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/7: send *ua.PublishRequest with 112 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: readChunk EOF
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: close
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: Close()
2024-01-19T14:57:41+08:00 E! [telegraf] Error running agent: starting input inputs.opcua_listener: failed to start monitoring items: EOF
```
Is there currently a solution to this problem?

jminardi, 2025-03-21:
@yang750418 did you ever figure out a solution to your problem? I am seeing the same behavior.

magiconair, 2025-03-22:
Closing this one and moved the other issue from @yang750418 to #781.

closed 2025-03-22 by magiconair