gopcua/opcua #661: Retry sync client calls with usual errors

bitomaxsp, 2023-06-09:
Problem: monitor loop handles reconnects and session reestablishing gracefully depending on the result of the last operation. 
But syn call result (error in this case) is also propagated back to the caller (which is user of the library who calls sync methods). Some of those errors could be retried internally and user request could be resend. 

Usual suspects that we at Northvolt retry are:
```go
func readErrorProc(err error, timeout *uint) bool {
	switch {
	case errors.Is(err, io.EOF): // connection closed by server
		*timeout = 0
		return true

	case errors.Is(err, ua.StatusBadSecureChannelIDInvalid): // secure channel is renewing
		*timeout = 0
		return true

	case errors.Is(err, ua.StatusBadSessionNotActivated):
		*timeout = 0
		return true

	case errors.Is(err, ua.StatusBadSessionIDInvalid):
		*timeout = 0
		return true

	case errors.Is(err, ua.StatusBadServerNotConnected):
		*timeout = 0
		return true

	case errors.Is(err, ua.StatusBadTimeout):
		*timeout += 1
		return true
	}

	return false
}

```

For us it is dramatically improved stability of the application working over OPCUA.

Going forward we could probably have configurable retry logic for synchronous client calls. Wdyt @magiconair and @kung-foo ?

sruehl, 2023-08-16:
Not sure, but this might be a related issue: https://github.com/gopcua/opcua/issues/239

sruehl, 2023-08-16:
Just as an option because it has a really simple API: https://github.com/avast/retry-go

So the code samples from OP above can be nicely combined with that.

closed 2025-04-21 by github-actions[bot]