gopcua/opcua #678: StatusBadEncodingLimitsExceeded on single read request

dtq-two, 2023-08-23:
After connecting to an OPC UA Server by "Softing dataFEED" I did a simple ReadRequest which resulted in a StatusBadEncodingLimitsExceeded.

The request was done like this:

```go
        // ... create client and connect
	_, err = client.ReadWithContext(ctx, &ua.ReadRequest{
		NodesToRead: []*ua.ReadValueID{{
			NodeID:      ua.MustParseNodeID("ns=0;i=2269"),
			AttributeID: ua.AttributeIDValue,
		}},
	})
	if err != nil {
		return fmt.Errorf("read with context: %w", err)
	}
```

Debugging showed that the error was created at here: https://github.com/gopcua/opcua/blob/807c2dabf202632c5fcb5c6ec00f182f3b97063c/ua/variant.go#L117-L146

The []byte that was decoded looked like this:

```
[]uint8 len: 17, cap: 65478, [140,255,255,255,255,96,227,5,236,109,213,217,1,255,255,255,255]
```

which resulted in an `m.arrayLength` of `-1`

### Looking at the Node-ID in UaExpert

[image]

I'm unsure how I could work around this error or fix this issue. Another problem is that the `client.State()` becomes `Disconnected` and `Reconnecting` after this issue without returning back to `Connected`.

magiconair, 2023-08-23:
Which gopcua version are you using?

Which data structure did you use for the `Read` response?

Regarding `state`: lets try to recreate the issue and then we can probably see where it breaks.

dtq-two, 2023-08-23:
> Which gopcua version are you using?

I was using version v0.3.5

> Which data structure did you use for the Read response?

What do you mean? In version v0.3.5 the `ReadWithContext` returns `(*ua.ReadResponse, error)`, but I'm unsure if this is what you meant.

magiconair, 2023-08-23:
OK, could you capture a tcpdump and attach this here? If you don't want to attach it here you can send it to me via Keybase.

magiconair, 2023-08-23:
I'm having trouble accessing my Keybase account right now. I might be able to get to it later tonight.

dtq-two, 2023-08-23:
> OK, could you capture a tcpdump and attach this here? If you don't want to attach it here you can send it to me via Keybase.

```
13:36:24.439016 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [S], seq 3640104227, win 65535, options [mss 1460,nop,wscale 6,nop,nop,TS val 791976457 ecr 0,sackOK,eol], length 0
13:36:24.439541 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [S.], seq 3279600835, ack 3640104228, win 65535, options [mss 1460,nop,wscale 1,sackOK,TS val 42226121 ecr 791976457], length 0
13:36:24.439624 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 1, win 2058, options [nop,nop,TS val 791976457 ecr 42226121], length 0
13:36:24.439745 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [P.], seq 1:102, ack 1, win 2058, options [nop,nop,TS val 791976457 ecr 42226121], length 101
13:36:24.440260 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [P.], seq 1:29, ack 102, win 32717, options [nop,nop,TS val 42226121 ecr 791976457], length 28
13:36:24.440353 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 29, win 2058, options [nop,nop,TS val 791976458 ecr 42226121], length 0
13:36:24.440507 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [P.], seq 102:234, ack 29, win 2058, options [nop,nop,TS val 791976458 ecr 42226121], length 132
13:36:24.440992 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [P.], seq 29:164, ack 234, win 32651, options [nop,nop,TS val 42226121 ecr 791976458], length 135
13:36:24.441076 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 164, win 2056, options [nop,nop,TS val 791976458 ecr 42226121], length 0
13:36:24.441328 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [P.], seq 234:542, ack 164, win 2056, options [nop,nop,TS val 791976458 ecr 42226121], length 308
13:36:24.442739 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 164:1612, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976458], length 1448
13:36:24.442740 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 1612:3060, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976458], length 1448
13:36:24.442809 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 1612, win 2033, options [nop,nop,TS val 791976461 ecr 42226121], length 0
13:36:24.442812 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 3060, win 2011, options [nop,nop,TS val 791976461 ecr 42226121], length 0
13:36:24.443187 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 3060:4508, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976461], length 1448
13:36:24.443188 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 4508:5956, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976461], length 1448
13:36:24.443190 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 5956:7404, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976461], length 1448
13:36:24.443191 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 7404:8852, ack 542, win 32497, options [nop,nop,TS val 42226121 ecr 791976461], length 1448
13:36:24.443238 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 4508, win 2025, options [nop,nop,TS val 791976461 ecr 42226121], length 0
13:36:24.443240 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 5956, win 2002, options [nop,nop,TS val 791976461 ecr 42226121], length 0
13:36:24.443368 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 8852, win 2025, options [nop,nop,TS val 791976461 ecr 42226121], length 0
13:36:24.443589 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 8852:10300, ack 542, win 32497, options [nop,nop,TS val 42226122 ecr 791976461], length 1448
13:36:24.443595 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 10300:11748, ack 542, win 32497, options [nop,nop,TS val 42226122 ecr 791976461], length 1448
13:36:24.443597 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 11748:13196, ack 542, win 32497, options [nop,nop,TS val 42226122 ecr 791976461], length 1448
13:36:24.443601 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], seq 13196:14644, ack 542, win 32497, options [nop,nop,TS val 42226122 ecr 791976461], length 1448
13:36:24.443607 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [P.], seq 14644:14927, ack 542, win 32497, options [nop,nop,TS val 42226122 ecr 791976461], length 283
13:36:24.443672 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 10300, win 2025, options [nop,nop,TS val 791976461 ecr 42226122], length 0
13:36:24.443798 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 14927, win 2020, options [nop,nop,TS val 791976461 ecr 42226122], length 0
13:36:24.443922 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [P.], seq 542:698, ack 14927, win 2048, options [nop,nop,TS val 791976462 ecr 42226122], length 156
13:36:24.444363 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [P.], seq 14927:15023, ack 698, win 32419, options [nop,nop,TS val 42226122 ecr 791976462], length 96
13:36:24.444401 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 15023, win 2046, options [nop,nop,TS val 791976462 ecr 42226122], length 0
13:36:24.444539 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [P.], seq 698:829, ack 15023, win 2048, options [nop,nop,TS val 791976462 ecr 42226122], length 131
13:36:24.444819 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [P.], seq 15023:15322, ack 829, win 32354, options [nop,nop,TS val 42226122 ecr 791976462], length 299
[... cut ...]
13:36:24.445608 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [F.], seq 957, ack 15404, win 2048, options [nop,nop,TS val 791976463 ecr 42226122], length 0
13:36:24.445822 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [.], ack 958, win 32290, options [nop,nop,TS val 42226122 ecr 791976463], length 0
13:36:24.445822 IP 10.10.20.11.4982 > 10.10.10.66.54401: Flags [F.], seq 15404, ack 958, win 32290, options [nop,nop,TS val 42226122 ecr 791976463], length 0
13:36:24.445883 IP 10.10.10.66.54401 > 10.10.20.11.4982: Flags [.], ack 15405, win 2048, options [nop,nop,TS val 791976463 ecr 42226122], length 0

```

magiconair, 2023-08-24:
Thanks. Can you attach the file? I need to look at the content of the response to understand where it breaks.

dtq-two, 2023-08-24:
> Thanks. Can you attach the file? I need to look at the content of the response to understand where it breaks.

This zip contains the tcpdump pcap file:

[read.pcap.zip](https://github.com/gopcua/opcua/files/12429898/read.pcap.zip)

commit 199ccf6e (2024-12-09): Issue #678: Fix Variant to handle nil slices

Fix the Variant encoder and decoder to handle nil slices
correctly by setting the length to -1.

Closes #678
commit 3c34d35e (2024-12-10): Issue #678: Fix Variant to handle nil slices

Fix the Variant encoder and decoder to handle nil slices
correctly by setting the length to -1.

Closes #678
PR #755 (2024-12-11): Issue #678: Fix Variant to handle nil slices
Fix the Variant encoder and decoder to handle nil slices
correctly by setting the length to -1.

https://reference.opcfoundation.org/Core/Part6/v104/docs/5.2.5

Closes #678
closed 2024-12-11 by magiconair