gopcua/opcua #679: id_gen.go too large

sruehl, 2023-08-23:
at the moment id_gen.go is a pretty big file. Github has trouble rendering it and my IDE (Goland) just bails because the file is too big (it even can't resolve symbols so you get red errors everywhere).

[image]
[image]

Looking into it I had the idea to split the file up into 1+n files where n is the number of `NodeClass`es ([ref](https://reference.opcfoundation.org/Core/Part6/v104/docs/A.3)). So we would have on `common.go` for the `Name` function feeding from n maps and a `id_${NodeClass}_gen.go` file for nodes grouped by class.

so codewise nothing would change as the name map is private anyway at the moment, we would just split up the load between multiple files.

sruehl, 2023-08-23:
I named `common.go` to `id_names_gen.go` in the PR

sruehl, 2023-08-29:
On note on this PR. The variables enum is still to big. In plc4x I solved it with this regex `^([A-Z][a-z]*)(.*),Variable$` -> `$1$2,Variable$1` https://github.com/apache/plc4x/commit/b2f3acb97b3b209b0434c2441bdb09a9eee641b6. I could extend this PR to split them even more up like I did in the plc4x commit.

kung-foo, 2023-08-31:
I'm kinda on the fence about this. On one hand, I don't like adding complexity and code paths specifically to address IDE and web interface deficiencies, on the other hand, I also use Goland and have hit this issue before (though I "solved" it by bumping up Goland's limits). @magiconair thoughts?

magiconair, 2023-10-23:
Hmm, this works just fine in VSCode and neovim. Isn't this a limit that you can tune in Goland? Seems like an arbitrary limit.

I could live with splitting the enums and the map into two different files but making a random split because of a Goland limit seems pretty weird. Let's try to fix the IDE first. 

[image]

magiconair, 2023-10-23:
I also don't follow the code-generation limit argument. The number of enum values does not change. So why would splitting into multiple files not hit that limit?

Also, since the code generator is java you should be able to just provide more memory with `-Xmx` or not? 

I don't think we had such a situation before and I'm also a bit on the fence here. On the one hand this code should just work. On the other hand I don't know how many people actually have this problem and whether the ones who do haven't just updated their settings and moved on. That's what I would do at least. 

We can add something to the README if there is a simple workaround. If there isn't and the library is otherwise unusable with either tool then we should fix it.

magiconair, 2023-10-23:
If we have to split it up then I'd suggest we do this by id, e.g.

```
id_00000_gen.go
id_10000_gen.go
id_20000_gen.go
...
id_names_gen.go
```

magiconair, 2023-11-29:
@sruehl is this still relevant?

magiconair, 2023-11-29:
I've just asked a friend to check this and he says that there is a button to enable the syntax highlighting with the suggestions to bump the memory and the instructions to do so.

magiconair, 2024-12-10:
@sruehl lets just fix this. I think I was making a silly argument here. Lets try to unblock you and others who run into that problem. Can you rebase your PR and I'll merge it. Thank you.

closed 2024-12-10 by magiconair