gopcua/opcua #682: Problem with using ReferencedNodes

sruehl, 2023-08-24:
When using `ReferencedNodes` new Nodes will be created using the `NodeID` of a `ExpandedNode`. This `NodeID` has flags for namespace and server which (the values as we use Node not ExpandedNode) then are not passed when using those Nodes
```golang
// ReferencedNodes returns the nodes referenced by this node.
func (n *Node) ReferencedNodes(ctx context.Context, refs uint32, dir ua.BrowseDirection, mask ua.NodeClass, includeSubtypes bool) ([]*Node, error) {
	if refs == 0 {
		refs = id.References
	}
	var nodes []*Node
	res, err := n.References(ctx, refs, dir, mask, includeSubtypes)
	if err != nil {
		return nil, err
	}
	for _, r := range res {
		nodes = append(nodes, n.c.Node(r.NodeID.NodeID))
	}
	return nodes, nil
}
```
here you can see the problematic line: `nodes = append(nodes, n.c.Node(r.NodeID.NodeID))`

So I have a server which responds with that on a `BrowseRequest` call:
`UA Secure Conversation Message: BrowseRequest`
[image]
When using then those nodeIDs for read it produces a invalid request because of the flags and the server returns a error:
`UA Secure Converation Message[Malformed Packet]`
[image]
here in the request above you can see that the flags `c0` are set from the retrieved `ExpandedNode`
`ErrorMessage`
[image]

sruehl, 2023-08-24:
@magiconair could it be that this [todo](https://github.com/gopcua/opcua/blob/main/ua/node_id.go#L16) is the root cause?
node_id.go:16 "`// todo(fs): fix mask`"

sruehl, 2023-08-24:
https://github.com/gopcua/opcua/pull/683 fixes the issue

kung-foo, 2023-08-28:
@sruehl is there overlap with the issue described here? https://github.com/gopcua/opcua/pull/619

sruehl, 2023-08-28:
I think it is basically the same Problem. Although I like the way #619 is a bit more efficient I would prefer having a `NodeFromExpandedNode` factory function which copies the values and set's the mask like it is done in #619. The function there called `RemoveExpandedFlag` doesn't make much sense outside the copy scenario. My approach does that externals so the `NodeFromExpandedNode` would be more efficient because it has access to the private fields. I could update the PR with that suggestion if wanted.

sruehl, 2023-08-28:
my approach == #683

closed 2023-08-29 by kung-foo