gopcua/opcua #698: EOF when connecting via secure connection

JeremyTheocharis, 2023-12-05:
Hi,

We are the maintainer of benthos-umh (https://github.com/united-manufacturing-hub/benthos-umh), which uses gopcua.

We found that connecting via a secure connection will to some servers closing the connection. We were even able to reproduce it with the Microsoft OPC UA simulator (https://github.com/united-manufacturing-hub/benthos-umh/blob/master/tests/docker-compose.yaml).

What is happening is we get an EOF from the server. When using no security, we can successfully connect. The server returns:
```

```

We have double checked the certificate generation, key sizes, etc., but could not find anything odd https://github.com/united-manufacturing-hub/benthos-umh/blob/bdd75d410d90e732641769cc81cae2bf63c4f898/plugin/opcua.go#L336-L458.

These are our client and server logs:
```
=== RUN   TestAgainstSimulator/Logging_Endpoints
    opcua_test.go:154: Endpoint 1:
    opcua_test.go:155:   EndpointURL: opc.tcp://ba26b088a1b0:50000/
    opcua_test.go:156:   SecurityMode: MessageSecurityModeSignAndEncrypt
    opcua_test.go:157:   SecurityPolicyURI: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
    opcua_test.go:158:   TransportProfileURI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
    opcua_test.go:159:   SecurityLevel: 106
    opcua_test.go:163:   Server ApplicationURI: urn:OpcPlc:ba26b088a1b0
    opcua_test.go:164:   Server ProductURI: https://github.com/azure-samples/iot-edge-opc-plc
    opcua_test.go:165:   Server ApplicationName: OpcPlc
    opcua_test.go:166:   Server ApplicationType: ApplicationTypeServer
    opcua_test.go:167:   Server GatewayServerURI:
    opcua_test.go:168:   Server DiscoveryProfileURI:
    opcua_test.go:169:   Server DiscoveryURLs: [opc.tcp://ba26b088a1b0:50000/]
    opcua_test.go:788:   Server certificate:
    opcua_test.go:805:     Not Before: 2023-11-13 00:00:00 +0000 UTC
    opcua_test.go:806:     Not After: 2024-11-13 00:00:00 +0000 UTC
    opcua_test.go:807:     DNS Names: [ba26b088a1b0]
    opcua_test.go:808:     IP Addresses: []
    opcua_test.go:809:     URIs: [urn:OpcPlc:ba26b088a1b0]
    opcua_test.go:184:   UserIdentityToken 1:
    opcua_test.go:185:     PolicyID: 1
    opcua_test.go:186:     TokenType: UserTokenTypeAnonymous
    opcua_test.go:187:     IssuedTokenType:
    opcua_test.go:188:     IssuerEndpointURL:
    opcua_test.go:184:   UserIdentityToken 2:
    opcua_test.go:185:     PolicyID: 2
    opcua_test.go:186:     TokenType: UserTokenTypeUserName
    opcua_test.go:187:     IssuedTokenType:
    opcua_test.go:188:     IssuerEndpointURL:
    opcua_test.go:184:   UserIdentityToken 3:
    opcua_test.go:185:     PolicyID: 3
    opcua_test.go:186:     TokenType: UserTokenTypeCertificate
    opcua_test.go:187:     IssuedTokenType:
    opcua_test.go:188:     IssuerEndpointURL:
    opcua_test.go:154: Endpoint 2:
    opcua_test.go:155:   EndpointURL: opc.tcp://ba26b088a1b0:50000/
    opcua_test.go:156:   SecurityMode: MessageSecurityModeSignAndEncrypt
    opcua_test.go:157:   SecurityPolicyURI: http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep
    opcua_test.go:158:   TransportProfileURI: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
    opcua_test.go:159:   SecurityLevel: 108
    opcua_test.go:163:   Server ApplicationURI: urn:OpcPlc:ba26b088a1b0
    opcua_test.go:164:   Server ProductURI: https://github.com/azure-samples/iot-edge-opc-plc
    opcua_test.go:165:   Server ApplicationName: OpcPlc
    opcua_test.go:166:   Server ApplicationType: ApplicationTypeServer
    opcua_test.go:167:   Server GatewayServerURI:
    opcua_test.go:168:   Server DiscoveryProfileURI:
    opcua_test.go:169:   Server DiscoveryURLs: [opc.tcp://ba26b088a1b0:50000/]
    opcua_test.go:788:   Server certificate:
    opcua_test.go:805:     Not Before: 2023-11-13 00:00:00 +0000 UTC
    opcua_test.go:806:     Not After: 2024-11-13 00:00:00 +0000 UTC
    opcua_test.go:807:     DNS Names: [ba26b088a1b0]
    opcua_test.go:808:     IP Addresses: []
    opcua_test.go:809:     URIs: [urn:OpcPlc:ba26b088a1b0]
    opcua_test.go:184:   UserIdentityToken 1:
[... 239 lines cut ...]
[10:05:21 ERR] TCPSERVERCHANNEL ForceChannelFault Socket=01734046, ChannelId=0, TokenId=0, Reason=BadSecurityChecksFailed 'Could not verify security on OpenSecureChannel request.'
[10:05:21 INF] ChannelId 36: in Faulted state.
[10:05:21 ERR] TCPSERVERCHANNEL ForceChannelFault Socket=01734046, ChannelId=0, TokenId=0, Reason=BadConnectionClosed 'Remote side closed connection'
[10:05:21 ERR] TCPSERVERCHANNEL ForceChannelFault Socket=01A0A231, ChannelId=32, TokenId=1, Reason=BadConnectionClosed 'Remote side closed connection'
[10:05:21 INF] ChannelId 32: in Faulted state.

This is what fails on the server:
// verify signature.
                if (!rsa.VerifyData(dataToVerify.Array, dataToVerify.Offset, dataToVerify.Count, signature, algorithm, padding))
                {
                    string messageType = Encoding.UTF8.GetString(dataToVerify.Array, dataToVerify.Offset, 4);
                    int messageLength = BitConverter.ToInt32(dataToVerify.Array, dataToVerify.Offset + 4);
                    string actualSignature = Utils.ToHexString(signature);
                    Utils.LogError("Could not validate signature.");
                    Utils.LogCertificate(LogLevel.Error, "Certificate: ", signingCertificate);
                    Utils.LogError("MessageType ={0}, Length ={1}, ActualSignature={2}",
                        messageType, messageLength, actualSignature);
                    return false;
                }
rsa.VerifyData is System.Security.Cryptography.X509Certificates.X509Certificate2.VerifyData
```

Is it because of our implementation of gopcua, or might it be an underlying issue in the library? We are glad about any help.

We even tried setting the certificate of the server to the proper IP address / DNS name (https://github.com/Azure-Samples/iot-edge-opc-plc/issues/318), but even this did not fix it. Because we are receiving this from various servers out there (namely OpenPCS7 from Siemens, B&R), we assume that this has something to do with either our implementation or the gopcua library

magiconair, 2023-12-06:
Hmm, it says `Certificate Validation failed. Reason=BadCertificateUntrusted`. 

I vaguely remember that there were some weird issues with padding. I assume you've done all the obvious stuff, like marking the certificate as trusted in the server.

The other thing you could try is to create a root certificate and sign a client certificate with that. Then import the root cert into the server as a trusted authority to see if that changes anything. 

I am not the expert on the crypto code. @dwhutchison do you maybe have an idea?

JeremyTheocharis, 2023-12-06:
Thank you for your reply! I assume it has nothing to do with trusting the certificate, as other clients as UAExpert and other tools can access the server without their certificate being needed to trust.

We could try your approach with the root certificate, maybe even let it create with a different tool. With this we could check if there is a problem with the certificate generation, or if its with the secure connection.

@dwhutchison If you have any other pointers for us, that would be great!

JeremyTheocharis, 2024-04-02:
I think the reason was quite stupid: it happened when we were creating the certificate as always, but then selecting a authentication method that was unsupported. We still don;t know if it is a client or server issue, however swtiching to "Basic256Sha256" works. 

Also another side issue: the error returned from the server is only available when debugging, but is overwritten at some point in the code with EOF. If anyone in the future has that problem, just open a debugger and fetch the proper error message

closed 2024-04-02 by JeremyTheocharis