gopcua/opcua #716: Trouble with calling NewClient()

justnat3, 2024-03-27:
Hey, i've been running into something quite strange (running `go 1.22.1 windows/amd64`)

I've tried this on linux, and a couple of different versions of go, I get the same result. I also tried creating a certificate on my machine, and within go using crypto/x509.

I inlined the code from setCertificate() which works. However when creating a new client it only tells me that I have a "malformed certificate" which to me looks like the same code. 

Is there something I am doing wrong here?

```go
func main() {

	// creates a valid x509 certificate, and a private key
	cert, priv := GenerateCertWithKey() // ([]byte, *rsa.PrivateKey)

	// this succeeds, the same code called in (gocpua)setCertificate()
	parsedCert, err := x509.ParseCertificate(cert)
	if err != nil {
		log.Fatalf("Failed to parse certificate: %s", err)
		return
	}

	// Produces the correct output
	log.Println(parsedCert.URIs)

	ctx := context.Background()

	// This fails parsing the certificate
	c, err := opcua.NewClient(
		Endpoint,
		opcua.SecurityMode(ua.MessageSecurityModeSignAndEncrypt),
		opcua.SecurityPolicy("Basic256Sha256"),
		opcua.PrivateKey(priv),
		opcua.Certificate(cert),
	)

	if err != nil {
		log.Fatal(err)
	}

	defer c.Close(ctx)
	err = c.Connect(ctx)
	if err != nil {
		log.Fatal(err)
	}
}
```

Dipp3r, 2024-06-19:
Hi,
I tried reproducing this issue in my environment and found that the error("malformed certificate") you are receiving is completely unrelated to the certificate you are generating. You may have to add other options as part of your client instance `opcua.SecurityFromEndpoint(ep, ua.UserTokenTypeAnonymous)` and `opcua.AuthAnonymous()` if you are trying to set an anonymous connection to the server.

Below are the client logs with and without these options:

```
// your case

opts := []opcua.Option{
  opcua.SecurityMode(ua.MessageSecurityModeSignAndEncrypt),
  opcua.SecurityPolicy("Basic256Sha256"),
  opcua.CertificateFile(certFile),
  opcua.PrivateKeyFile(keyFile),
}
```

> 2024/06/19 22:49:01 ❌ Error while setting up a connection "opc.tcp://localhost:port/server-name"
> 2024/06/19 22:49:01 x509: malformed certificate
> exit status 1

```
// with authType set to anonymous

opts := []opcua.Option{
  opcua.SecurityMode(ua.MessageSecurityModeSignAndEncrypt),
  opcua.SecurityPolicy("Basic256Sha256"),
  opcua.SecurityFromEndpoint(ep, ua.UserTokenTypeAnonymous),    // ep is the endpointDescription returned by the SelectEndpoint method
  opcua.AuthAnonymous(),
  opcua.CertificateFile(certFile),
  opcua.PrivateKeyFile(keyFile),
}
```

> 2024/06/19 22:48:22 ✅ Connected to the server with endpoint "opc.tcp://localhost:port/server-name"
> exit status 1

hbrackel, 2025-02-11:
Hello,

a little late to the topic, but I encountered the same problem. The error is about the remote server's certificate, which the client thinks is not correct. The only way I found to work in this case is to first read the server's endpoints and select the one with the desired security scheme. Then use the `opcua.SecurityFromEndpoint()` option to configure the client

```
// read endpoints from server
// select the one with the desired security (see example https://github.com/gopcua/opcua/blob/main/examples/endpoints/endpoints.go )

opt := opcua.SecurityFromEndpoint(selectedEndpointDescription, id.AnonymousIdentityToken)

```

magiconair, 2025-02-13:
How could we improve the client to either make this more obvious or not have the error in the first place? Are there better defaults we could set?

hbrackel, 2025-02-13:
I only started using gopcua for about 2 weeks now - maybe a little early to provide meaningful advice. What I would wish is something like the following:

1. When *creating* a client,  I only want to configure the "static" security scheme (MessageSecurityMode, SecurityPolicy, Auth) and timeouts, retries,  etc without the need to discover the server's endpoints. At the time I create the client I may not yet be able to connect to the server yet
2. The specification says that the client can decide to accept differences between the server's certificate and what the server transmits during the opening of the session; one such thing might be a different applicationUri in the cert from what it communicates during the session opening. To make this decision at connection time, the client could call a configured callback with the server cert properties and some of the communicated props as input params so that the callback can accept or reject the connection/certificate/...
3. The current interaction with gopcua is relatively low level - e.g. using ReadRequest and Responses. It is excellent that this is possible, but for regular coding of standard tasks (such as reading a variable value, calling a method) some convenience wrappers might hide some of the details.

For the first and last bullet point, I started to write a small *UaClient* wrapper, in which you can just *seemingly* create and configure a *UaClient*, but under the hood, the actual gopcua client is only created on call of Connect(), which first executes GetEndpoints and then creates the client with using the endpoints properties. This also allows to defer the provision of the endpointUrl until you really connect.
Again - I just started to discover gopcua's capabilities and I certainly did not yet fully grasp all of its features and. I am using a commercial Java SDK though for more than 12 years.

closed 2025-08-21 by github-actions[bot]