gopcua/opcua #723: Get rid of dependencies

michelseiler, 2024-04-25:
It would be nice if this package reduce the dependencies.

Package `github.com/pkg/errors` is used in `errors/errors.go`:
- `pkg_errors.Errorf(Prefix+format, a...)` could be replaced with `fmt.Errorf(Prefix+format, a...)`
- `pkg_errors.New(Prefix + text)` could be replaced with `fmt.Errorf(Prefix + text)`

Package `golang.org/x/exp` is used in `client_sub.go` and `debug/debug.go` for slices:
- It could perhaps be replaced by the standard library function [slices.Contains()](https://pkg.go.dev/slices#Contains) when Bumping GO version to `1.21`.

closed 2024-06-13 by kung-foo

michelseiler, 2024-06-13:
Thanks! 
Should i write another issue for the `golang.org/x/exp` package since this is not resolved by PR https://github.com/gopcua/opcua/pull/731?

kung-foo, 2024-06-13:
As I wrote (just now) in https://github.com/gopcua/opcua/pull/721, we are going to hold at Go 1.20 due to compatibility issues which means the `slices.Contains` generic function isn't available.

michelseiler, 2024-06-13:
Thanks for the clarification