gopcua/opcua #738: Example / Method for browsing nodes to get the child node and its details

ekafe, 2024-07-17:
I am trying to read a node that is having a children of 1000-2000 tags, I need to get the children node name and node class , so that i can show it in my ui, currently i am using the below method to get the children 

```
children, err := client.Node(nodeID).Children(context.Background(), 0, ua.NodeClassAll)
for _, child := range children {
  nodeType, _ := child.NodeClass(context.Background())
  displayName, _ := child.DisplayName(context.Background())
}
```

i have used this method to get the data, but then i noticed that nodes having more children is taking more time to get the request.
then only i understood i am making seperate calls to opcua server for child node class and display names.

i have checked the code 
```children, err := client.Node(nodeID).Children(context.Background(), 0, ua.NodeClassAll)``` 
this only gives node Id , not any details about the node
```
        {
                "ID": "ns=2;s=Channel1.Device1.tag830"
        },
        {
                "ID": "ns=2;s=Channel1.Device1.tag831"
        },
```

i also tried this code 
```refs, err := client.Node(nodeID).ReferencedNodes(context.Background(), 0, ua.BrowseDirectionForward, ua.NodeClassAll, true)```
 this also get only the nodeId, not nodeclass or displayname
```
       {
                "ID": "ns=2;s=Channel1.Device1.tag830"
        },
        {
                "ID": "ns=2;s=Channel1.Device1.tag831"
        },
```

then i saw the one method called browse, which i couldn't find any example, i have tried the below commands to browse the node and get children
```
	var browserDescs []*ua.BrowseDescription
	browseDescription := &ua.BrowseDescription{
		NodeID:          nodeID,
		ReferenceTypeID: ua.NewNumericNodeID(0, id.References),
		BrowseDirection: ua.BrowseDirection(ua.BrowseDirectionForward),
		IncludeSubtypes: false,
		NodeClassMask:   1,
		ResultMask:      1,
	}

	browserDescs = append(browserDescs, browseDescription)

	browseRequest := ua.BrowseRequest{
		View: &ua.ViewDescription{
			ViewID: ua.NewTwoByteNodeID(0),
		},
		NodesToBrowse: browserDescs,
	}
	response, err := client.Browse(context.Background(), &browseRequest)
	if err != nil {
		log.Printf("error browsing : %v", err)
	}
```
In response i am not getting any details about the nodes or children.

I want to verify, if the above browse code is correct or not, if I am missed or written wrongly ?
I also want to know I am using the browse method correctly, is it for browsing nodes or not.?

Is it not possible to read the children node with details with current library implementation, or if am missing any simple method to do this.

jackchenjc, 2024-08-22:
Maybe you can try to use [Node.References](https://pkg.go.dev/github.com/gopcua/opcua#Node.References) and handle `[]*ua.ReferenceDescription` in your code.
It is also be used in [Node.ReferenceNodes](https://github.com/gopcua/opcua/blob/708f669296f58deba4d100507115e02166c0f0f9/node.go#L158) which only [return the nodeId](https://github.com/gopcua/opcua/blob/708f669296f58deba4d100507115e02166c0f0f9/node.go#L163) as you mentioned.

ekafe, 2024-08-23:
thanks for help, yes i have gone through the code
```
children_ref, err := client.Node(nodeID).References(context.Background(), id.HierarchicalReferences, ua.BrowseDirectionForward, ua.NodeClassAll, true)
```

then i loop through the ref and got the id and browse name

closed 2024-08-23 by ekafe