gopcua/opcua #750: [Question] How to connect opcua server By UaExpert

Ri0nGo, 2024-11-15:
The NodeSet2_server and node_server packages in the example directory start a server program, and I use UaExpert to connect to the server, and the connection fails, is there any configuration that needs to be added?

This UaExpert connect config
[image]
[image]
Then, I click ignore, log is BadConfigurationError.
[image]

Looking forward to your reply, many thanks!

danomagnum, 2024-11-17:
Seems to work for me.  Can you do a wireshark capture while trying to connect?

If possible, could you also try connecting with python and let me know how that goes?

```
>>> import opcua
cryptography is not installed, use of crypto disabled
cryptography is not installed, use of crypto disabled
>>> client = opcua.Client('opc.tcp://localhost:4840')
>>> client.connect()
Requested session timeout to be 3600000ms, got 0ms instead
>>> root = client.get_root_node()
>>> root.get_children()
[Node(TwoByteNodeId(i=85)), Node(TwoByteNodeId(i=86)), Node(TwoByteNodeId(i=87))]
>>> 
```

Ri0nGo, 2024-11-20:
Sorry for taking so long to get back to you. I reinstalled UaExpert and it connects fine.

Ri0nGo, 2024-11-20:
solved

closed 2024-11-20 by Ri0nGo