gopcua/opcua #768: Method output arguments return <nil> Value() for ExtensionObject arrays representing custom structure dataTypes

hbrackel, 2025-02-11:
When calling a method whose output argument[0] is an array of custom structure dataType values, the corresponding output argument returns a slice of `[]*ua.ExtensionObject`, where the `ua.ExtensionObject.Value() == nil` for each slice element. The same method called from e.g. UaExpert returns the correct decoded custom structure; that is the method returns the correct []*ua.ExtensionObject. Tested with multiple clients on multiple custom OPC UA servers. As a result, the ExtensionObject value cannot be decoded.

pseudo code:
```
...
oID := objectNodeId
mID := methodNodeId // method with zero input arguments; one output argument
req := &ua.CallMethodRequest{
		ObjectID:       oID,
		MethodID:       mID,
		InputArguments: []*ua.Variant{},
	}

resp, err := c.Call(ctx, req)
if err != nil {
	log.Fatal(err)
}
if got, want := resp.StatusCode, ua.StatusOK; got != want {
	log.Fatalf("got status %v want %v", got, want)
}
out := resp.OutputArguments[0].Value() 
// first output argument shall be an array of values of a custom structure dataType; 
// thus encoded as extensionObjects

extObjects, ok := out.([]*ua.ExtensionObject) // works up to here; 
// ...error handling omitted
for _, extObj range extObjects {
  value := extObj.Value(). // value is always nil??
}

```

hbrackel, 2025-02-11:
in this concrete case, the method returns a single output argument, where the arg's variant is expected to contain an array of ExtensionObjects (encoded structured dataType values).

So: correct

hbrackel, 2025-02-11:
@magiconair the signatures of the test are correct, but the output arguments would best be encoded *custom OPC UA structure dataType* values, while the test uses go structure types.
For a 100% reproduction of the problem, the server should would  a custom OPC UA structure dataType (none of the  OPC UA predefined structures) and return an array of values of that structured OPC UA dataType.

I don't know whether the problem is specifically about receiving encoded custom structure ExtensionObjects or just "any" ExtensionObjects. 

Thanks for taking the time and effort!

magiconair, 2025-02-12:
I thought that the `Complex` data type is a custom data type encoded as an ExtensionObject but maybe I'm mistaken. I'll have a look at the nodeset.

```python
class Complex(uatypes.FrozenClass):
    ua_types = [
        ('i', 'Int64'),
        ('j', 'Int64'),
    ]

    def __init__(self, i=0, j=0):
        self.i = i
        self.j = j
        self._freeze = True

    def __str__(self):
        return f'Complex(i:{self.i}, j:{self.j})'

    __repr__ = __str__
...

if __name__ == "__main__":
    server = Server()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/")

    ns = server.register_namespace("http://gopcua.com/")
    uatypes.register_extension_object('Complex', ua.NodeId("ComplexType", ns), Complex)
...
```

magiconair, 2025-02-12:
Could you capture a tcpdump of the UAExpert method call and attach it here?

Maybe something like this:

```shell
tcpdump -n -i eth0 -s 1520 -w dump.pcap port 4840
```

hbrackel, 2025-02-12:
The other difference is that ComplexNumberType is an OPC UA defined type (namespace 0, known to gopcua, while the custom structure is not. I am not familiar enough with Python OPC UA to predict what the ua.Variant([Complex(1,2), Complex(3,4]) will be encoded to. (line 41 in your test)

magiconair, 2025-02-12:
OK. If you could send me the tcpdump file then I can see what the protocol looks on the wire and see why the code doesn't decode it into an array of extension objects with the `Issue768DataType` from your nodeset.

hbrackel, 2025-02-13:
I hope the attached *.pcab is what you are looking for. Given limited admin rights on my company notebook, I created a server from the nodeset on my personal macOS machine. The recording is from a method call of the `issue768` method using ProsysOPC's "Browser" client. The call succeeded and returned a 2 element array of the properly decoded `Issue768DataType` values. I did not try to call the method using the gopcua client (which is on the company computer). So I hope that this is not a platform related issue.

[dump.pcap.zip](https://github.com/user-attachments/files/18784714/dump.pcap.zip)

magiconair, 2025-02-14:
In gopcua you need to register the extension objects before you can use them. I think you are used to Java where the client SDK discovers the type from the server and dynamically creates a corresponding class. Try adding this to your code:

```go
	type Issue768DataType struct {
		Name       string
		NodeId     *NodeID
		LowerBound time.Time
		UpperBound time.Time
	}

	RegisterExtensionObject(NewNumericNodeID(2, 5001), new(Issue768DataType))
```

I've written a test case and hand-decoded the byte array and this works:

```go
func TestIssue768(t *testing.T) {
	type Issue768DataType struct {
		Name       string
		NodeId     *NodeID
		LowerBound time.Time
		UpperBound time.Time
	}

	RegisterExtensionObject(NewNumericNodeID(2, 5001), new(Issue768DataType))

	data := []byte{
		// variant

		// encoding mask: VariantArrayValues  (0x80) | TypeIDExtensionObject (0x16)
		0x96,

		// array length
		0x01, 0x00, 0x00, 0x00,

		// ExtensionObject[0]

		// TypeID
		// NodeIDTypeFourByte (0x01)
		0x01,
		// Namespace: 2
		0x02,
		// identifier: 5001
		0x89, 0x13,

		// EncodingMask: Binary (0x01)
		0x01,

		// Length: 43
		0x2b, 0x00, 0x00, 0x00,

		// Value

		// Name
		// length: 8
		0x08, 0x00, 0x00, 0x00,
		// value: Issue768
		0x49, 0x73, 0x73, 0x75, 0x65, 0x37, 0x36, 0x38,

		// NodeId
		// type: string
		0x03,
		// namespace: 123
		0x7b, 0x00,
		// lentgth: 8
		0x08, 0x00, 0x00, 0x00,
		// value: myNodeId
		0x6d, 0x79, 0x4e, 0x6f, 0x64, 0x65, 0x49, 0x64,

		// LowerBound
		0xf0, 0x49, 0x58, 0x0f, 0x1c, 0x7e, 0xdb, 0x01,

		// UpperBound
		0xf0, 0x49, 0x58, 0x0f, 0x1c, 0x7e, 0xdb, 0x01,
	}

	t.Run("codec", func(t *testing.T) {
		ts := 0x01DB7E1C0F5849F0
		now := time.Unix(0, int64((ts-116444736000000000)*100)).UTC()

		cases := []CodecTestCase{
			{
				Name: "[]*ua.ExtensionObject",
				Struct: MustVariant([]*ExtensionObject{
					NewExtensionObject(&Issue768DataType{
						"Issue768",
						NewStringNodeID(123, "myNodeId"),
						now,
						now,
					}),
				}),
				Bytes: data,
			},
		}
		RunCodecTest(t, cases)
	})

	t.Run("decode", func(t *testing.T) {
		var v Variant
		_, err := Decode(data, &v)
		require.NoError(t, err)

		val := v.Value().([]*ExtensionObject)
		require.Len(t, val, 1)
		t.Logf("val[0]: %#v", val[0].Value)
	})
}
```

magiconair, 2025-02-14:
Since Go is a compiled language the current implementation needs to pre-register all extensions object types at compile time. You need to use the Node ID for the binary encoding to register it (2, 5001) in your case. 

So far this has worked mostly for us but I've since written some code which dynamically creates a type at runtime using reflection to support decoding. This is only useful for serializing it to JSON IMHO or some other form with reflection since you don't have access to the fields. 

We use this internally to decode generic UDTs from PLCs and forward them as JSON to an API. There is a discussion thread about this. (https://github.com/gopcua/opcua/discussions/441)

magiconair, 2025-02-14:
It's in the examples but maybe easy to miss

https://github.com/gopcua/opcua/blob/main/examples/udt/udt.go#L78-L97

hbrackel, 2025-02-14:
Once again, thanks a lot for explaining this. 
I am actually connecting to an aggregating server, which in turn dynamically forwards and "aggregates" underlying servers. This presents 2 challenges: UDTs can only be discovered at runtime, not compile time, and the absolute NodeIds of types are not predictable, only ExpandedNodeIds (the namespaceArray is growing as new server's are connected). My idea was to dynamically (on demand) read the DataTypeDefinition attribute of a UDT (possibly recursively) and then decode an extension object based on that knowledge. This would require that the extensionObject at least contains the UDT values as an opaque byte array, without the need to pre-register the nodes.

I would like to avoid the need to pre-register UDTs as there are literally hundreds  of them. Where in the code is the ExtensionObject's payload of non-registered UDTs set to nil? If an unregistered UDTs ExtensionObject's Value() would provide the raw payload bytes, I may be able to decode "manually".

magiconair, 2025-02-14:
OK, that's what I thought. We compiled the structures into our mapping applications since at some point we need to understand the schema anyway and have access to the actual fields and data types.

One could argue that you can write a small app which discovers the types, generates the Go code including the registration call and stores them in a schema repo that you just import instead of dealing with `map[string]any` and converting to JSON only. That's not how Go usually works. Bit of a mind shift if you come from another language.

The approach is the one outlined in https://github.com/gopcua/opcua/discussions/691#discussioncomment-7058178. It would work well as a starting point for Go type discovery and generation as well as dynamic handling of types. 

Maybe it is time to generalize this approach and add it to the library to support both use cases. You could help develop/test it if you want. Let me see if there is a good starting point. 

Contributions usually start with fork and PR and over time a small number of people have taken on larger portions of the code (@kung-foo as general maintainer, @danomagnum for the server, @dwhutchison for the initial crypto code) and these people have more access to the repo, create branches without forking etc. Especially, for the server code which is fairly new and maintained by @danomagnum.

magiconair, 2025-02-14:
> I would like to avoid the need to pre-register UDTs as there are literally hundreds of them. Where in the code is the ExtensionObject's payload of non-registered UDTs set to nil? If an unregistered [...]

There is no code that sets it to `nil`. It is `nil` because the code knows that it is an Array of Extension Objects (0x96) with length 1 and type id 2,5001 but has no corresponding type definition to decode it. So it never gets set. 

It should probably return either an error or have access to the raw payload. However, if you are that far you might as well define a Go type and let the decoder do the work.

magiconair, 2025-02-14:
As a former Java programmer with now 14 years of Go under the belt I remember that it felt really nice to go back to engineering principles and hand code certain things because it isn't really that much work. Sometimes it just seems like it.

hbrackel, 2025-02-14:
> There is no code that sets it to `nil`. It is `nil` because the code knows that it is an Array of Extension Objects (0x96) with length 1 and type id 2,5001 but has no corresponding type definition [...]

This was actually one of my questions: how can I access the raw payload of the extension object? I thought `Value` would be delivering it, but here is where I get `nil`.

I actually don't need to convert it back into the original Go structure rather only need a sort map[key]value per dataType value instance for further e.g. JSON export processing.

Back on Tuesday - on the road now.

hbrackel, 2025-02-17:
Hi,

after registering the DataTypeEncoding ID the method returns the proper array of UDTs.

Thanks for your patience and support. I will close this issue then with the comment.

I would have a couple of suggestions on this topic, which I probable create separate Github issues for:
a) the registration of ExtensionObjects could support ExpandedNodeIds, as the actual namespaceIndex may only be available after connecting to the server (I am connecting to an aggregating server with a non-static namespaceArray)
b) the registration should be "per client". I am creating multiple clients connecting to multiple servers within in the same application . Each server could have a different UDT for the same DataTypeEncodingId
c) as you already suggested, if the client can't find a registered encoding, the extension object could contain the raw payload []byte data
d) if the previous would be available, one could implement dynamic decoding into a map[any]any for

I could implement the last bullet point, but would certainly need quite some guidance on the 3rd one.

Cheers, HAns-Uwe

PR #770 (2025-04-08): Add test for method call with no inputs and an array of extobjs as output
Closes #768
commit f7b9bfb6 (2025-04-08): tests/python: add test for issue #768

Call method without input args and return a single value with an
array of extension objects.

Fixes #768
closed 2025-02-17 by hbrackel