gopcua/opcua #772: The value.nodeid method returned by the read function returns Nil

mijjjj, 2025-02-13:
The value.nodeid method returned by the read function returns Nil, which makes it impossible for me to locate the value I get back after sending multiple nodeids.

magiconair, 2025-02-13:
I would need a bit more context to understand the problem. Did you check the `StatusCode` if that was `OK`? Does the node you are reading have a NodeID as value data type?

Did you have a look at https://github.com/gopcua/opcua/blob/main/examples/read/read.go ?

mijjjj, 2025-02-13:
I just used the corresponding code in the example. The specific code is as follows. req.ReqData contains multiple nodeids. When reading the results, the nodeid obtained through resp.Results[0].Value.NodeID() is nil

[image]

mijjjj, 2025-02-13:
At present, when I want to determine which nodeid it is, I can only judge it by the array address.

hbrackel, 2025-02-13:
Could you please provide your ReadRequest, in particular the `NodesToRead` array? 

If you are requesting attribute values from multiple NodeIds, the ReadResponse provides an array of DataValues (one per requested NodeId and Attribute). None of the DataValues contains information about the NodeId or AttributeId from which it had been read, so you need to associate the returned DataValues with the `NodeToRead` by it position (aka array index). This is how OPC UA standard wants it to be and is not related to gopcua.