gopcua/opcua #774: Read panicked with nil pointer dereference

FingersXed, 2025-02-27:
Hello everyone,
after upgrading from v0.5.3 to v0.7, the following panic occurred after a series of failed read/write attempts (EOF, StatusBadTimeout). The laptop was in stand-by at that time and continuously losing its WIFI connection.

```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x68 pc=0xea0489]

goroutine 104 [running]:
github.com/gopcua/opcua/uasc.(*SecureChannel).SendRequestWithTimeout(0x0, {0x1aad1d0, 0xc000446700}, {0x1aa7fc0, 0xc001212ba0}, 0xc000d72540, 0x2540be400, 0xc0012c7860)
	/go/pkg/mod/github.com/gopcua/opcua@v0.7.0/uasc/secure_channel.go:979 +0x49
github.com/gopcua/opcua.(*Client).sendWithTimeout(0x25ed1a0?, {0x1aad1d0?, 0xc000446700?}, {0x1aa7fc0?, 0xc001212ba0?}, 0x41bd65?, 0x8?)
	/go/pkg/mod/github.com/gopcua/opcua@v0.7.0/client.go:959 +0x132
github.com/gopcua/opcua.(*Client).Send(0xc00074c0e0, {0x1aad1d0, 0xc000446700}, {0x1aa7fc0, 0xc001212ba0}, 0xc0012c7860)
	/go/pkg/mod/github.com/gopcua/opcua@v0.7.0/client.go:941 +0x88
github.com/gopcua/opcua.(*Client).Read(0xc00074c0e0, {0x1aad1d0, 0xc000446700}, 0xc0012128d0)
	/go/pkg/mod/github.com/gopcua/opcua@v0.7.0/client.go:1048 +0x28c
****************************/pkg/nuopc.Read({0x1aad1d0, 0xc000446700}, 0xc00074c0e0, {0x15ef2e0?, 0xc001286900?}, 0xc0012c7b88?)
	/app/pkg/nuopc/opc.go:36 +0x268
****************************/core/plc.(*readQO).Execute(0xc001284e10, 0xc00074c0e0)
	/app/core/plc/queue_object.go:71 +0x93
****************************/core/plc.(*PLC).run(0xc0008b2840, 0xc00090fbe0)
	/app/core/plc/plc.go:402 +0x4b2
****************************/core/plc.InitShared.func1 in goroutine 1
	/app/core/plc/plc.go:66 +0x2e7
```

This has only happened once since the update, and I couldn't manage to recreate it (tried by repeatedly turning WIFI off/on). Any help would be much appreciated.

Thanks in advance and best regards.

magiconair, 2025-02-27:
Looks like we need to do the `nil` check more than once.

https://github.com/gopcua/opcua/blob/main/client.go#L950-L960

magiconair, 2025-02-27:
Actually, maybe we need to use the nil-checked secure channel I think. Not sure why we don't. Maybe a merge bug.

magiconair, 2025-02-27:
yes, this is a regression introduced in commit 4d56b5d. The race condition was fixed in 5a15559.

FingersXed, 2025-02-27:
Thank you kind sir!

PR #775 (2025-02-27): client: use nil-checked secure channel
this is a regression introduced in commit https://github.com/gopcua/opcua/commit/4d56b5d38e0759d0a9782df6e1984e6c9b41a956. The race condition was fixed in https://github.com/gopcua/opcua/commit/5a155594a69dc78ee664d28607a89f4e3b97ceed.

Fixes #774
commit c58424ae (2025-03-16): client: use nil-checked secure channel

This fixes a regression from commit 4d56b5d
which reverts the fix for the race condition
added in 5a15559

Fixes #774