gopcua/opcua #776: Writable variable in server

think-free, 2025-03-05:
I don't manage to have a writable variable in the server
I create the node like this :

`		ua.AttributeIDAccessLevel:             server.DataValueFromValue(accessLevel),
		ua.AttributeIDUserAccessLevel:         server.DataValueFromValue(accessLevel),`

Creating the server like this : 

`	opts := []server.Option{
		server.EnableSecurity("None", ua.MessageSecurityModeNone),
		server.EnableAuthMode(ua.UserTokenTypeAnonymous),
		server.EndPoint("0.0.0.0", 4840),
		server.EndPoint("localhost", 4840),
	}
	s := server.New(opts...)`

Using FreeOpcUA GUI client it gives me the following error : 

asyncua.ua.uaerrors._auto.BadUserAccessDenied: User does not have permission to perform the requested operation.(BadUserAccessDenied)

I've tried lot of things without success

How can I make a variable writable and knowing that the client have changed it ?

danomagnum, 2025-03-06:
I'm not sure about the access level stuff.  I'll take a look at it this evening.

To know that a client has changed a variable, I made a custom namespace (meeting the `namespace` interface  https://github.com/gopcua/opcua/blob/a5114dafc2e0bf4035badb6d46448aaa2ce104d6/server/namespaces.go#L18 ) and added my event logic into the SetAttribute function.  You could effectively copy the NodeNamespace.go file from the server and rename it, adding whatever you want there.

I admit if you just want to get notified this may be somewhat cumbersome (but then again this is OPCUA, so everything is cumbersome).  I could potentially add a ChangeNotification channel in the server that would allow it to propogate the nodeID of any node that was changed out of the server if needed.

That would be done here if it is a worthwhile idea.  Any input there would be welcome.
https://github.com/gopcua/opcua/blob/a5114dafc2e0bf4035badb6d46448aaa2ce104d6/server/server.go#L189-L191

danomagnum, 2025-03-06:
I see that when using access controls, if it exists, it is just denying.  I'll try to get the actual checks implemented.  Shouldn't be too difficult.

danomagnum, 2025-03-07:
Try using `go get github.com/gopcua/opcua@server_accesslevels` and see if it works now.  

Also, see the updated access examples in the node server: https://github.com/gopcua/opcua/blob/5a715188011c360825c5a9690f55886d4a28d020/examples/server/node_server/server.go#L212-L252

think-free, 2025-03-10:
Hi !
It's fixed :) 
Thank you very much for your quick response and fix

magiconair, 2025-04-07:
The access level code has now been merged into v0.7.4. Closing

closed 2025-04-07 by magiconair