gopcua/opcua #781: EOF in readChunk when reading 1500 nodes

magiconair, 2025-03-22:
Moved from https://github.com/gopcua/opcua/issues/658#issuecomment-1903473439

@magiconair Hello

I use Telegraf OPC UA plugin, which uses gopcua.
I am currently encountering the same situation. Using security and OPCUA to subscribe, EOF occurs when Nodes exceeds 1000 or 1500.
the logs..
```
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Connecting OPC UA Client to server
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp: connecting to opc.tcp://192.168.1.XX:48050
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: start HEL/ACK handshake
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: sent HELF with 60 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv ACKF with 28 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv OPNF with 136 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: recv OPNF with 136 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: recv *ua.OpenSecureChannelResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: received security token. channelID=2437948123 tokenID=1 createdAt=2024-01-19T06:57:35Z lifetime=1h0m0s
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: security token is refreshed at 2024-01-19T07:42:41Z (45m0s). channelID=2437948123 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: security token expires at 2024-01-19T08:12:35Z. channelID=2437948123 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: send *ua.GetEndpointsRequest with 97 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: recv MSGF with 3690 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: recv MSGF with 3690 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: recv *ua.GetEndpointsResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/2: sending *ua.GetEndpointsResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: Close()
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 1: close
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 1: readChunk EOF
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Configuring OPC UA connection options
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Loading cert/key from /etc/telegraf/ua.pem//etc/telegraf/ua_key.pem
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] security policy from configuration 
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp: connecting to opc.tcp://192.168.1.XX:48050
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: start HEL/ACK handshake
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: sent HELF with 60 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv ACKF with 28 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0xffff00, MaxChunkCount:0x100}
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: send *ua.OpenSecureChannelRequest with 1905 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv OPNF with 1901 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: setting securityPolicy to http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: recv OPNF with 1901 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: recv *ua.OpenSecureChannelResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/1: sending *ua.OpenSecureChannelResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: received security token. channelID=2437948124 tokenID=1 createdAt=2024-01-19T06:57:35Z lifetime=1h0m0s
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: security token is refreshed at 2024-01-19T07:42:41Z (45m0s). channelID=2437948124 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: security token expires at 2024-01-19T08:12:35Z. channelID=2437948124 tokenID=1
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: send *ua.CreateSessionRequest with 1584 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 5408 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: recv MSGF with 5408 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: recv *ua.CreateSessionResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/2: sending *ua.CreateSessionResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: send *ua.ActivateSessionRequest with 464 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 144 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: recv MSGF with 144 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: recv *ua.ActivateSessionResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/3: sending *ua.ActivateSessionResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/4: send *ua.ReadRequest with 144 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 480 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/4: recv MSGF with 480 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/4: recv *ua.ReadResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/4: sending *ua.ReadResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Connected to OPC UA Server
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Creating OPC UA subscription
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/5: send *ua.CreateSubscriptionRequest with 128 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: recv MSGF with 112 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/5: recv MSGF with 112 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/5: recv *ua.CreateSubscriptionResponse
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/5: sending *ua.CreateSubscriptionResponse to handler
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] Subscribed with subscription ID 2432947172
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65536 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/6: send *ua.CreateMonitoredItemsRequest with 20544 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2/7: send *ua.PublishRequest with 112 bytes
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: readChunk EOF
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uacp 2: close
2024-01-19T14:57:41+08:00 D! [inputs.opcua_listener] uasc 2: Close()
2024-01-19T14:57:41+08:00 E! [telegraf] Error running agent: starting input inputs.opcua_listener: failed to start monitoring items: EOF
```
Is there currently a solution to this problem?

jminardi, 2025-03-24:
I am seeing the same issue:

```log
2025-03-24T16:51:48Z W! Running without secret protection!
2025-03-24T16:51:48Z I! Loading config: /etc/telegraf/telegraf.conf
2025-03-24T16:51:48Z I! Loading config: /etc/telegraf_configs/config-0.conf
2025-03-24T16:51:48Z I! Starting Telegraf 1.33.1 brought to you by InfluxData the makers of InfluxDB
2025-03-24T16:51:48Z I! Available plugins: 236 inputs, 9 aggregators, 33 processors, 26 parsers, 63 outputs, 6 secret-stores
2025-03-24T16:51:48Z I! Loaded inputs: internal opcua statsd
2025-03-24T16:51:48Z I! Loaded aggregators:
2025-03-24T16:51:48Z I! Loaded processors: enum
2025-03-24T16:51:48Z I! Loaded secretstores:
2025-03-24T16:51:48Z I! Loaded outputs: influxdb
2025-03-24T16:51:48Z I! Tags enabled: host=insights-telegraf-0
2025-03-24T16:51:48Z I! [agent] Config: Interval:10s, Quiet:false, Hostname:"insights-telegraf-0", Flush Interval:10s
2025-03-24T16:51:48Z W! [agent] The default value of 'skip_processors_after_aggregators' will change to 'true' with Telegraf v1.40.0! If you need the current default behavior, please explicitly set the option to 'false'!
2025-03-24T16:51:48Z D! [agent] Initializing plugins
2025-03-24T16:51:48Z D! [inputs.opcua] Initialising OpcUAInputClient
2025-03-24T16:51:48Z D! [inputs.opcua] Initialising OpcUAClient
2025-03-24T16:51:48Z D! [inputs.opcua] Initialising node to metric mapping
2025-03-24T16:51:48Z D! [agent] Connecting outputs
2025-03-24T16:51:48Z D! [agent] Attempting connection to [outputs.influxdb]
2025-03-24T16:51:48Z D! [agent] Successfully connected to outputs.influxdb
2025-03-24T16:51:48Z D! [agent] Starting service inputs
2025-03-24T16:51:48Z I! [inputs.statsd] UDP listening on "0.0.0.0:8125"
2025-03-24T16:51:48Z I! [inputs.statsd] Started the statsd service on ":8125"
2025-03-24T16:51:49Z D! [inputs.opcua] Connecting OPC UA Client to server
2025-03-24T16:51:49Z D! [inputs.opcua] uacp: connecting to opc.tcp://ts2-kep-01.test.ihies.com:49320
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: start HEL/ACK handshake
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: sent HELF with 73 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: recv ACKF with 28 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x1000000, MaxChunkCount:0x1388}
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: recv OPNF with 135 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/1: recv OPNF with 135 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/1: recv *ua.OpenSecureChannelResponse
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1: received security token. channelID=4019013706 tokenID=1 createdAt=2025-03-24T16:51:55Z lifetime=1h0m0s
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1: security token is refreshed at 2025-03-24T17:36:49Z (45m0s). channelID=4019013706 tokenID=1
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1: security token expires at 2025-03-24T18:06:55Z. channelID=4019013706 tokenID=1
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/2: send *ua.GetEndpointsRequest with 110 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: recv MSGF with 1924 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/2: recv MSGF with 1924 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/2: recv *ua.GetEndpointsResponse
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/2: sending *ua.GetEndpointsResponse to handler
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1: Close()
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 1: close
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 1: readChunk EOF
2025-03-24T16:51:49Z D! [inputs.opcua] Configuring OPC UA connection options
2025-03-24T16:51:49Z D! [inputs.opcua] Loading cert/key from /etc/telegraf/cert.pem//etc/telegraf/key.pem
2025-03-24T16:51:49Z D! [inputs.opcua] security policy from configuration http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
2025-03-24T16:51:49Z D! [inputs.opcua] User cares about both the policy (http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256) and security mode (MessageSecurityModeSignAndEncrypt)
2025-03-24T16:51:49Z D! [inputs.opcua] Server has 1 endpoints
2025-03-24T16:51:49Z D! [inputs.opcua] Evaluating endpoint opc.tcp://TS2-KEP-01.test.ihies.com:49320, policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256, mode MessageSecurityModeSignAndEncrypt, level 51
2025-03-24T16:51:49Z D! [inputs.opcua] Security policy and mode found. Using server endpoint opc.tcp://TS2-KEP-01.test.ihies.com:49320 for security. Policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
2025-03-24T16:51:49Z D! [inputs.opcua] uacp: connecting to opc.tcp://ts2-kep-01.test.ihies.com:49320
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 2: start HEL/ACK handshake
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 2: sent HELF with 73 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 2: recv ACKF with 28 bytes
2025-03-24T16:51:49Z D! [inputs.opcua] uacp 2: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0xffff, SendBufSize:0xffff, MaxMessageSize:0x1000000, MaxChunkCount:0x1388}
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 2/1: send *ua.OpenSecureChannelRequest with 1647 bytes
[... cut ...]
debug: sub: pause: pause
2025-03-24T16:51:49Z E! [inputs.opcua] Error in plugin: registering nodes failed: write tcp 10.42.0.26:39958->10.5.40.100:49320: use of closed network connection
2025-03-24T16:51:49Z D! [inputs.opcua] uasc 2: Close()
```

I also have a packet capture file I cam share if you think that will be helpful.

I do not get this EOF error if the number of nodes is around 1500 or less. I also do not get this error if I am not using encryption. (security_policy=None and security_mode=None)

jminardi, 2025-03-26:
I have been digging into this with the kepware support team. We enabled OPCUA logs on the kepware runtime. The gopcua logs and kepware logs matched 1:1 up until the `ua.RegisterNodesRequest`. That request does not show up in the kepware logs. Digging into network captures, we can see that the kepware runtime is rejecting it immediately before any sort of processing can happen. The theory is that the request size is too large. Notice the number of bytes is 65536. All the logs were sent to the kepware engineering team and they are digging into it.

When running with encryption disabled, you can see the message size is set to `65534`.

@magiconair Any idea why the message size is larger when using encryption? Is there any way to get it 2 bytes smaller?

Logs with encryption disabled:
```
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/5: send *ua.RegisterNodesRequest with 1268 bytes
[inputs.opcua]  conn.go:384 uacp 2: recv MSGF with 42424 bytes
[inputs.opcua]  secure_channel.go:350 uasc 2/5: recv MSGF with 42424 bytes
[inputs.opcua]  secure_channel.go:289 uasc 2/5: recv *ua.RegisterNodesResponse
[inputs.opcua]  secure_channel.go:304 uasc 2/5: sending *ua.RegisterNodesResponse to handler
[inputs.opcua]  secure_channel_instance.go:89 got sequence number 15
[inputs.opcua]  secure_channel.go:1053 uasc 2/6: send *ua.ReadRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/6: send *ua.ReadRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/6: send *ua.ReadRequest with 59714 bytes
[inputs.opcua]  conn.go:384 uacp 2: recv MSGC with 65535 bytes
[inputs.opcua]  secure_channel.go:350 uasc 2/6: recv MSGC with 65535 bytes
[inputs.opcua]  conn.go:384 uacp 2: recv MSGC with 65535 bytes
[inputs.opcua]  secure_channel.go:350 uasc 2/6: recv MSGC with 65535 bytes
[inputs.opcua]  conn.go:384 uacp 2: recv MSGC with 65535 bytes
[inputs.opcua]  secure_channel.go:350 uasc 2/6: recv MSGC with 65535 bytes
[inputs.opcua]  conn.go:384 uacp 2: recv MSGF with 23143 bytes
[inputs.opcua]  secure_channel.go:350 uasc 2/6: recv MSGF with 23143 bytes
[inputs.opcua]  secure_channel.go:289 uasc 2/6: recv *ua.ReadResponse
[inputs.opcua]  secure_channel.go:304 uasc 2/6: sending *ua.ReadResponse to handler
[inputs.opcua]  secure_channel_instance.go:89 got sequence number 18
[inputs.opcua]  secure_channel.go:1053 uasc 2/7: send *ua.ReadRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/7: send *ua.ReadRequest with 65534 bytes
[inputs.opcua]  secure_channel.go:1053 uasc 2/7: send *ua.ReadRequest with 59714 bytes
```

magiconair, 2025-04-15:
@jminardi could you check if #792 improves the situation? I'm a bit out of my depth with encryption algorithms and padding but I think I'm looking at the right spot. If this addresses the behavior then I'll invest more time to fully understand the implementation.

reindlt, 2025-04-29:
Hi, I experience the same issue with Telegraf's `opcua_listener` plugin.

I have built a version with the fix applied and tested it together with the latest version of Telegraf which is using v0.7.4. It seems that with the fix applied it no longer works with encryption disabled but at least there is another `CreateMonitoredItemsRequest`.

Here are the relevant parts of the logs:

Log with fix applied with encryption enabled:
```
...
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65552 bytes
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 62464 bytes
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 10
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:329 uasc 2: readChunk EOF
2025-04-29T06:41:33Z E! [telegraf] Error running agent: starting input inputs.opcua_listener::opcua-au: failed to start monitoring items: EOF
debug: client: monitor: disconnected
debug: client: monitor: auto-reconnecting
debug: client: monitor: action: createSecureChannel
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  conn.go:217 uacp 2: close
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1180 uasc 2: Close()
debug: client: monitor: trying to recreate secure channel
2025-04-29T06:41:33Z D! [inputs.opcua_listener::opcua-au]  conn.go:70 uacp: connecting to opc.tcp://127.0.0.1:4840/opcua
```

Log with v0.7.4 with encryption enabled:
```
...
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65536 bytes
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:329 uasc 2: readChunk EOF
debug: client: monitor: disconnected
debug: client: monitor: auto-reconnecting
debug: client: monitor: action: createSecureChannel
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  conn.go:217 uacp 2: close
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 8
2025-04-29T06:45:01Z E! [telegraf] Error running agent: starting input inputs.opcua_listener::opcua-au: failed to start monitoring items: write tcp 127.0.0.1:34839->127.0.0.1:4840: wsasend: An existing connection was forcibly closed by the remote host.
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1180 uasc 2: Close()
debug: client: monitor: trying to recreate secure channel
2025-04-29T06:45:01Z D! [inputs.opcua_listener::opcua-au]  conn.go:70 uacp: connecting to opc.tcp://127.0.0.1:4840/opcua
debug: publish: PublishResponse: null
```

Log with fix applied with encryption disabled:
```
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65538 bytes
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65538 bytes
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:329 uasc 2: readChunk EOF
debug: client: monitor: disconnected
debug: client: monitor: auto-reconnecting
debug: client: monitor: action: createSecureChannel
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  conn.go:217 uacp 2: close
2025-04-29T06:43:19Z E! [telegraf] Error running agent: starting input inputs.opcua_listener::opcua-au: failed to start monitoring items: write tcp 127.0.0.1:34599->127.0.0.1:4840: wsasend: An existing connection was forcibly closed by the remote host.
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 9
2025-04-29T06:43:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1180 uasc 2: Close()
debug: client: monitor: trying to recreate secure channel
```

Log with v0.7.4 with encryption disabled:
```
025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65534 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65534 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65534 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 62335 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 10
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/7: send *ua.PublishRequest with 66 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  conn.go:384 uacp 2: recv MSGF with 56364 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:350 uasc 2/6: recv MSGF with 56364 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:289 uasc 2/6: recv *ua.CreateMonitoredItemsResponse
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:304 uasc 2/6: sending *ua.CreateMonitoredItemsResponse to handler
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  conn.go:384 uacp 2: recv MSGF with 61831 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:350 uasc 2/7: recv MSGF with 61831 bytes
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au] Monitoring items
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:289 uasc 2/7: recv *ua.PublishResponse
2025-04-29T06:46:10Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:304 uasc 2/7: sending *ua.PublishResponse to handler
```

magiconair, 2025-04-29:
OK, I'll switch to the old formula and try to understand what's going on. @reindlt could you try a couple of more attempts? I'll update #792

reindlt, 2025-04-29:
Sure, I'm happy to help where I can

magiconair, 2025-04-29:
@reindlt I've pushed a change to #792. There was a suspicious off-by-one error in one of the formulas. Let's see if that works.

reindlt, 2025-04-30:
With encryption enabled it fails after the first `CreateMonitoredItemsRequest`:
```
025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65552 bytes
2025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:329 uasc 2: readChunk EOF
debug: client: monitor: disconnected
debug: client: monitor: auto-reconnecting
debug: client: monitor: action: createSecureChannel
2025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  conn.go:221 uacp 2: close
2025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1180 uasc 2: Close()
debug: client: monitor: trying to recreate secure channel
2025-04-30T07:31:19Z E! [telegraf] Error running agent: starting input inputs.opcua_listener::opcua-au: failed to start monitoring items: write tcp 127.0.0.1:21886->127.0.0.1:4840: use of closed network connection
2025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  conn.go:70 uacp: connecting to opc.tcp://127.0.0.1:4840/opcua
2025-04-30T07:31:19Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 8
```

With encryption disabled it fails after `PublishRequest`:
```
025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65538 bytes
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65538 bytes
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 65538 bytes
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/6: send *ua.CreateMonitoredItemsRequest with 62323 bytes
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel_instance.go:89 got sequence number 10
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1050 uasc 2/7: send *ua.PublishRequest with 66 bytes
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:329 uasc 2: readChunk EOF
debug: client: monitor: disconnected
debug: client: monitor: auto-reconnecting
debug: client: monitor: action: createSecureChannel
2025-04-30T07:35:03Z E! [telegraf] Error running agent: starting input inputs.opcua_listener::opcua-au: failed to start monitoring items: EOF
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  conn.go:221 uacp 2: close
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  secure_channel.go:1180 uasc 2: Close()
debug: publish: PublishResponse: null
debug: client: monitor: trying to recreate secure channel
debug: publish: eof: pausing publish loop
2025-04-30T07:35:03Z D! [inputs.opcua_listener::opcua-au]  conn.go:70 uacp: connecting to opc.tcp://127.0.0.1:4840/opcua
debug: sub: error: EOF
debug: sub: pause
debug: sub: pause: pause
```

magiconair, 2025-04-30:
Thank you. Could you capture a tcpdump of these two sessions and post them here?

reindlt, 2025-04-30:
[tcpdump.zip](https://github.com/user-attachments/files/19975982/tcpdump.zip)

reindlt, 2025-05-12:
@magiconair any news?

jminardi, 2025-09-04:
@sparkplow @reindlt I've pushed a fix for this issue that works for me here: ghcr.io/jminardi/telegraf:fix-781

Feel free to test it out and see if it fixes your issues, or view the PR at #824