gopcua/opcua #782: Regression when updating from 0.5.3 to 0.7.1

Scarjit, 2025-03-24:
After updating gopcua our test against a Siemens S7 and subscribing to a a struct fails.

Here is the relevent test code:
```go
	When("Subscribing to a struct", func() {
		It("should return data changes", func() {
			var err error

			var nodeIDStrings = []string{"ns=4;i=6"}

			parsedNodeIDs := ParseNodeIDs(nodeIDStrings)

			input = &OPCUAInput{
				OPCUAConnection: &OPCUAConnection{
					Endpoint:           endpoint,
					Username:           "",
					Password:           "",
					ServerCertificates: make(map[*ua.EndpointDescription]string),
				},
				NodeIDs:          parsedNodeIDs,
				SubscribeEnabled: false,
			}
			// Attempt to connect
			err = input.Connect(ctx)
			Expect(err).NotTo(HaveOccurred())

			// expect 2 messages for both nodes
			Eventually(func() (int, error) {
				messageBatch, _, err := input.ReadBatch(ctx)
				return len(messageBatch), err
			}, 30*time.Second, 100*time.Millisecond).WithContext(ctx).Should(Equal(2))
		})
	})
```
https://github.com/united-manufacturing-hub/benthos-umh/blob/master/opcua_plugin/opcua_plc_test.go#L211C8-L211C27

And this is the resulting failure:
```
• [FAILED] [45.354 seconds]
Test Against Siemens S7 [Serial]
/home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:75
  when Subscribing to a struct
  /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:211
    [It] should return data changes
    /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:212

Timeline
    > Enter [BeforeEach] Test Against Siemens S7 - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:83 @ 03/17/25 13:00:17.585
    < Exit [BeforeEach] Test Against Siemens S7 - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:83 @ 03/17/25 13:00:17.585 (0s)
    > Enter [It] should return data changes - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:212 @ 03/17/25 13:00:17.585
    [FAILED] Context was cancelled (cause: context deadline exceeded) after 24.946s.
    Expected
        <int>: 3
    to equal
        <int>: 2
  Error: It 03/17/25 13:01:02.753
  
    Full Stack Trace
      github.com/united-manufacturing-hub/benthos-umh/opcua_plugin_test.init.func8.5.1()
      	/home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:237 +0x2b7
    < Exit [It] should return data changes - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:212 @ 03/17/25 13:01:02.753 (45.168s)
    > Enter [AfterEach] Test Against Siemens S7 - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:96 @ 03/17/25 13:01:02.754
    < Exit [AfterEach] Test Against Siemens S7 - /home/runner/_work/benthos-umh/benthos-umh/opcua_plugin/opcua_plc_test.go:96 @ 03/17/25 13:01:02.939 (185ms)
```
https://github.com/united-manufacturing-hub/benthos-umh/actions/runs/13899936281/job/38888999092#step:10:4039

led0nk, 2025-03-25:
This is the struct we're using, which is exposed via opc-server-interface:
[image]

magiconair, 2025-04-15:
It looks like the problem is not in reading the struct but in the number of messages you get back. For some reason you are getting 3 instead of 2. Can you update the test to print the content of the messages? Is this a problem at all? Maybe your test is too strict. But lets find out what is going on here.