gopcua/opcua #783: server CreateSession does not return endpoint on trailing slash mismatch

bernhard-R, 2025-03-31:
Hi, I am in the process of setting up a opc ua server using for testing purposes of our .NET application.
Using this code to create the server 
```
var opts []server.Option

opts = append(opts, server.EnableSecurity("None", ua.MessageSecurityModeNone))

opts = append(opts,
	server.EnableAuthMode(ua.UserTokenTypeAnonymous),
)

opts = append(opts,
	server.EndPoint("localhost", 4840),
)

endpoints := []string{
	"localhost",
}
certbytes, keybytes, err := generateCert(endpoints, 4096, time.Minute*60*24*365*10)
if err != nil {
	log.Fatalf("problem creating cert: %v", err)
}

var cert []byte
c, err := tls.X509KeyPair(certbytes, keybytes)

if err != nil {
	log.Printf("Failed to load certificate: %s", err)
} else {
	pk, ok := c.PrivateKey.(*rsa.PrivateKey)
	if !ok {
		log.Fatalf("Invalid private key")
	}
	cert = c.Certificate[0]
	opts = append(opts, server.PrivateKey(pk), server.Certificate(cert))
}
	logger := Logger(1)
opts = append(opts,
	server.SetLogger(logger),
)
s := server.New(opts...)
return s
```

I am unable to create a session using our .NET client, as supposedly the endpoints do not match.
I have narrowed it down to the following code block in session_service.go

```
matching_endpoints := make([]*ua.EndpointDescription, 0)
for i := range s.srv.endpoints {
	ep := s.srv.endpoints[i]
	if ep.EndpointURL == req.EndpointURL {
		matching_endpoints = append(matching_endpoints, ep)
	}
}

response := &ua.CreateSessionResponse{
	ResponseHeader:        responseHeader(req.RequestHeader.RequestHandle, ua.StatusOK),
	SessionID:             sess.ID,
	AuthenticationToken:   sess.AuthTokenID,
	RevisedSessionTimeout: float64(sess.cfg.sessionTimeout / time.Millisecond),
	MaxRequestMessageSize: 0, // Not used
	ServerSignature: &ua.SignatureData{
		Signature: sig,
		Algorithm: alg,
	},
	ServerCertificate: s.srv.cfg.certificate,
	ServerNonce:       nonce,
	ServerEndpoints:   matching_endpoints,
}
```

The problem seems to be, that req.EndpointURL has a trailing slash i.e. `opc.tcp://localhost:4840/` (automatically appended by the .NET library) while the servers endpoint does not. (listing  s.Endpoints() gives me `opc.tcp://localhost:4840`)
So no endpoints are returned in the response.

Is this not matching a conscious decision / does the trailing slash matter? (I am no expert in working with opc and its endpoint-handling) or can I somehow configure the endpoint with a trailing slash?

Thanks!

magiconair, 2025-03-31:
I can have a look at this

danomagnum, 2025-04-09:
I don't see anything in the spec that says anything about trailing slashes.  I also don't see a reason the server should be picky about a trailing slash.

I added the code to ignore trailing slashes in the issue-783-trailing-slash branch.

https://github.com/gopcua/opcua/compare/main...issue-783-trailing-slash

If that sounds good enough, I'll make a PR.

bernhard-R, 2025-04-10:
This would definetely fix my issue :)

magiconair, 2025-04-10:
I'll release this once the PR is there.

PR #788 (2025-04-11): Server: Ignore trailing slashes when checking matching endpoints.
This pull request includes changes to the server endpoint results so that trailing slashes are ignored.

Fixes #783
closed 2025-04-11 by magiconair