gopcua/opcua #786: NodeId/ExpandedNodeId implementation mix API and decoder concerns

hbrackel, 2025-04-04:
The current implementation of the NodeId and ExpandedNodeIds types mixes the OPC UA Standard type definitions with decoding/encoding related concerns. This comes with several problems. 2 of those "problems" are
- There is no twoByte or fourByte NodeId. Only a single numeric one. (and Opaque, String, GUID) The number of bytes used for encoding/decoding is a concern of the en/decoder only and should not be part of the NodeId/ExpandedNodeId type. The user (aka developer) should not be exposed to two/four/single byte details
- none of the flags IndexFlag, UriFlag or Masks belongs in the type definitions or methods of the types (they are not necessary outside the Decoder/Encoder)

The conversion between NodeId and ExpandedNodeId required access to the serverArray and namespaceArray of the connected server. This implies that the conversion functions / Constructors require a connected client and cannot be used "offline" or standalone. A construct/type such as `AddressSpace` could be used to hold or relay the NamespaceArray, ServerArray and also the TypeDictionary (or the Client instance itself could be used). The AddressSpace would be created after a successful `Connect()`.

With the above, the current implementation could be enriched and become more usable and valuable

magiconair, 2025-04-05:
What are you suggesting? I read a couple of things here:

* Make the `NewNumericNodeId()` create the different encoding types (two byte, four byte, ...)
* Make the IndexFlag/URIFlag/Masks methods private or remove them altogether

I'm not sure what you mean with the conversion between NodeId and ExpandedNodeId. `ParseExpandedNodeId` takes a `[]string` which is the namespace array from the server. Isn't that enough or are you suggesting another helper method?

hbrackel, 2025-04-05:
I am currently updating the nodeid.go and expandednodeid.go files with the proposed changes. I may take another day or two. I will provide a link to the fork once this is done. I think it is easier to look at the code than trying to describe it.

magiconair, 2025-04-28:
No worries. We can always re-open. Just cleaning out some old stuff to keep it recent. Looking forward to your contribution! :)

closed 2025-11-02 by github-actions[bot]