gopcua/opcua #794: Decoding of non-registered custom structured datatypes returns nil ExtensionObject.Value

hbrackel, 2025-04-28:
When decoding a non-registered custom structured dataType, the ExtensionObject contains 'nil' as the Value component. We discussed in a previous conversation whether it would make sense to return the raw bytes of the value instead of nil.

Would this be as easy as changing the ExtensionObject.Decode as follows:

```go
func (e *ExtensionObject) Decode(b []byte) (int, error) {
	buf := NewBuffer(b)
	e.TypeID = new(ExpandedNodeID)
	buf.ReadStruct(e.TypeID)

	e.EncodingMask = buf.ReadByte()
	if e.EncodingMask == ExtensionObjectEmpty {
		return buf.Pos(), buf.Error()
	}

	length := buf.ReadUint32()
	if length == 0 || length == 0xffffffff || buf.Error() != nil {
		return buf.Pos(), buf.Error()
	}

	body := NewBuffer(buf.ReadN(int(length)))
	if buf.Error() != nil {
		return buf.Pos(), buf.Error()
	}

	if e.EncodingMask == ExtensionObjectXML {
		e.Value = new(XMLElement)
		body.ReadStruct(e.Value)
		return buf.Pos(), body.Error()
	}

	typeID := e.TypeID.NodeID
	e.Value = eotypes.New(typeID)
	if e.Value == nil {
		debug.Printf("ua: unknown extension object value dataType %s - returning raw byte array", typeID)
		raw := body.Bytes()
		e.Value = &raw
		return buf.Pos(), buf.Error()
	}

	body.ReadStruct(e.Value)
	return buf.Pos(), body.Error()
}
```

Would the raw bytes need to be cloned prior to returning them or is it safe as shown above?

magiconair, 2025-04-28:
Good question. This is testing my knowledge of Go internals. Good exercise to dig in a bit. I'll get back to you.

magiconair, 2025-04-28:
`Buffer` has a reference to the slice and `ReadN` also only returns slices which means that they all share the same underlying backing array. So yes, this would need to be cloned. Since go1.21 there is a function for this:

```go
e.Value = slices.Clone(body.Bytes())
```

Can you check if this works? This should be configurable though or we need to mark it as a breaking change.

hbrackel, 2025-04-28:
I tried both variants for a non-trivial custom dataType. The body.ReadStruct version returned 12 bytes whereas the one with body.Bytes() delivered > 100bytes. The latter makes more sense for the example at hand (although I did not try to manually decode the byte array yet).

hbrackel, 2025-04-29:
As for the configurability a boolean flag could be added to the TypeRegistry, indicating whether or not an empty <nil> payload shall be returned in case of an unknown UDT or the UDT's raw payload bytes. I'd prefer this flag being on the OPC UA client, but that's not accessible during the decoding.

magiconair, 2025-04-30:
Hmm, not sure I'd want to tack this onto the type registry. We could have a separate `DecoderConfig` and configure it via a `Config` option in the `Client` init. 

I think this points to a deeper problem since it makes the decoder behavior global. It'd be like having only one JSON decoder per application. You cannot have two clients to different servers with different ExtObj behavior in the same application. I think some of the design decisions from 6 years ago are finally catching up with me here 😁

Both `Encoder` and `Decoder` should be structs which have options. This is where the `RawBytes` option belongs IMHO. The question is how to pass that decoder down the chain without breaking all the signatures. 

If we'd configure decoder behavior as a client option then it could pass a decoder to the secure channel where it is used. I think this is where the start is.

magiconair, 2025-04-30:
The other problem is that `Buffer` and `Decoder`/`Encoder` depend on each other. `Buffer` should just be a helper and the `ReadStruct`/`WriteStruct` methods are a kludge ...

hbrackel, 2025-04-30:
I agree with the comments above; just the type Registry itself should also be „per client“, for the same arguments. In fact, I am using 2 clients in my current application, connecting to 2 different servers at the same time. This can create conflicting type registrations (same nodeId, different types).
With the ExtensionObjects now providing the raw value bytes, I did setup a custom TypeRegistry for each client, where, on demand, the DataTypeDefinition is recursively parsed and stored in the TypeRegistry. Next up is a kind of decoder, which iteratively decodes the ExtensionObject.Value bytes based on the Structure- or EnumDefinitions

magiconair, 2025-04-30:
I'm coming to the same conclusion and it might be as simple as this. Didn't think too hard about when `raw` should be populated on `Encode` or whether it should be populated at all. What do you think? Sounds like this is in line with what you have.

```patch
diff --git i/ua/extension_object.go w/ua/extension_object.go
index 4ee7daf..7271a48 100644
--- i/ua/extension_object.go
+++ w/ua/extension_object.go
@@ -5,6 +5,8 @@
 package ua
 
 import (
+	"slices"
+
 	"github.com/gopcua/opcua/debug"
 	"github.com/gopcua/opcua/id"
 )
@@ -35,13 +37,15 @@ const (
 type ExtensionObject struct {
 	EncodingMask uint8
 	TypeID       *ExpandedNodeID
-	Value        interface{}
+
+	raw   []byte
+	value interface{}
 }
 
 func NewExtensionObject(value interface{}) *ExtensionObject {
 	e := &ExtensionObject{
 		TypeID: ExtensionObjectTypeID(value),
-		Value:  value,
+		value:  value,
 	}
 	e.UpdateMask()
 	return e
@@ -66,21 +70,23 @@ func (e *ExtensionObject) Decode(b []byte) (int, error) {
 	if buf.Error() != nil {
 		return buf.Pos(), buf.Error()
 	}
+	e.raw = slices.Clone(body.Bytes())
 
 	if e.EncodingMask == ExtensionObjectXML {
-		e.Value = new(XMLElement)
-		body.ReadStruct(e.Value)
+		e.value = new(XMLElement)
+		body.ReadStruct(e.value)
 		return buf.Pos(), body.Error()
 	}
 
 	typeID := e.TypeID.NodeID
-	e.Value = eotypes.New(typeID)
-	if e.Value == nil {
+	e.value = eotypes.New(typeID)
+	if e.value == nil {
 		debug.Printf("ua: unknown extension object %s", typeID)
-		return buf.Pos(), buf.Error()
+		// return buf.Pos(), buf.Error()
+	} else {
+		body.ReadStruct(e.value)
 	}
 
-	body.ReadStruct(e.Value)
 	return buf.Pos(), body.Error()
 }
 
@@ -96,25 +102,38 @@ func (e *ExtensionObject) Encode() ([]byte, error) {
 	}
 
 	body := NewBuffer(nil)
-	body.WriteStruct(e.Value)
+	body.WriteStruct(e.value)
 	if body.Error() != nil {
 		return nil, body.Error()
 	}
+	e.raw = body.Bytes()
 	buf.WriteUint32(uint32(body.Len()))
 	buf.Write(body.Bytes())
 	return buf.Bytes(), buf.Error()
 }
 
 func (e *ExtensionObject) UpdateMask() {
-	if e.Value == nil {
+	if e.value == nil {
 		e.EncodingMask = ExtensionObjectEmpty
-	} else if _, ok := e.Value.(*XMLElement); ok {
+	} else if _, ok := e.value.(*XMLElement); ok {
 		e.EncodingMask = ExtensionObjectXML
 	} else {
 		e.EncodingMask = ExtensionObjectBinary
 	}
 }
 
+func (e *ExtensionObject) Bytes() []byte {
+	return e.raw
+}
+
+func (e *ExtensionObject) Value() interface{} {
+	return e.value
+}
+
+func (e *ExtensionObject) IsValid() bool {
+	return e.value != nil
+}
+
 func ExtensionObjectTypeID(v interface{}) *ExpandedNodeID {
 	switch v.(type) {
 	case *AnonymousIdentityToken:
```

magiconair, 2025-04-30:
> I agree with the comments above; just the type Registry itself should also be „per client“, for the same arguments. In fact, I am using 2 clients in my current application, connecting to 2 different [...]

Good point

hbrackel, 2025-05-04:
I like the `Bytes()` method always returning the raw ExtensionObject's bytes (if so desired). Different with the `IsValid()` method. The OPC UA standard describes that if a client does not understand the encoded DataType, it shall treat it as opaque data and skip over it . Both options (decoded or opaque data) are equally valid. So I would prefer to not implement the IsValid(). Or it may be named `IsDecoded()`. But because an ExtensionObject's payload cannot otherwise be `nil`, `nil` is sufficiently indicative IMO.

Making the `value`structure component private and introducing a corresponding getter `Value()` introduced 2 challenges: a) this really *is* a breaking change of the API and b) It introduces an inconsistency with the `TypeId` property, as this still can be directly modified. I am all for private properties plus accessors, but in that case all of the properties should be private, including `EncodingMask` and `TypeId`.

I suggest to separate the eventual rework / relocation of the TypeRegistry into a dedicated branch and not mix it with the discussed changes to the ExtensionObject.

hbrackel, 2025-05-08:
I tried to implement your proposed solution above. There are quite some other places to modify as well. Looking at the required changes, I think this might be a little "too much" for the intended purpose.

- the raw bytes are only meaningful for decoding, not for encoding (IMO); at least for the use cases I have in mind, a simple cloning into a separate property, such as `raw` would suffice
- the getter `Bytes()` on the extension object is attractive, but when using it in the actual code it might create some ambiguity whether the entire ExtensionObject's bytes are returned or just the "Value" bytes. Other names are very clunky, though (`RawValue()`, `ValueBytes()`, ... `raw`should also be a pointer, so that `nil` can be returned if the bytes have not been set.

closed 2025-11-12 by github-actions[bot]