gopcua/opcua #796: session timeout

HeczZots, 2025-04-30:
```

func main() {
	mctx := context.Background()
	ctx, cancel := context.WithCancel(mctx)
	opts := []opcua.Option{
		opcua.SecurityMode(ua.MessageSecurityModeNone),
		opcua.SecurityPolicy(ua.SecurityPolicyURINone),
		opcua.AutoReconnect(false),
		opcua.ReconnectInterval(5 * time.Second),
		opcua.RequestTimeout(10 * time.Second),
		opcua.SessionTimeout(120000),
	}

	client, err := opcua.NewClient(serverURL1, opts...)
	if err != nil {
		log.Fatal(err)
	}
	defer client.Close(ctx)

	if err := client.Connect(ctx); err != nil {
		log.Fatalf("Ошибка подключения: %v", err)
	}
}
```

I have receiving The operation timed out. StatusBadTimeout (0x800A0000)

[image]

this is UA expert packets

[image]

May be i can handle this error with some opcua.Options ?

HeczZots, 2025-04-30:
P.S. it doesnt metter which options im using, its never create a session at all

```
func (s *SecureChannel) sendRequestWithTimeout(
	ctx context.Context,
	req ua.Request,
	reqID uint32,
	instance *channelInstance,
	authToken *ua.NodeID,
	timeout time.Duration,
	h ResponseHandler) error {

	s.pendingReq.Add(1)
	respRequired := h != nil
	// i have received some data but session by no reason is not active
	ch, err := s.sendAsyncWithTimeout(ctx, req, reqID, instance, authToken, respRequired, timeout)
	s.pendingReq.Done()
	if err != nil {
		return err
	}

	if !respRequired {
		return nil
	}

	// `+ timeoutLeniency` to give the server a chance to respond to TimeoutHint
	timer := time.NewTimer(timeout + timeoutLeniency)
	defer timer.Stop()

	select {
	case <-ctx.Done():
		s.popHandler(reqID)
		return ctx.Err()
	case <-s.disconnected:
		s.popHandler(reqID)
		return io.EOF
	case msg := <-ch:
		if msg.Err != nil {
			if msg.Response() != nil {
				_ = h(msg.Response()) // ignore result because msg.Err takes precedence
			}
			return msg.Err
		}
		return h(msg.Response())
		// here is my error
	case <-timer.C:
		s.popHandler(reqID)
		return ua.StatusBadTimeout
	}
}
```

magiconair, 2025-05-01:
What's the content of the CreateSessionResponse?

HeczZots, 2025-05-01:
> What's the content of the CreateSessionResponse?
```

0000   4d 53 47 46 ce 01 00 00 01 00 00 00 01 00 00 00
0010   02 00 00 00 02 00 00 00 01 00 d0 01 80 db ab fe
0020   df b1 9d 01 02 00 00 00 00 00 00 00 00 ff ff ff
0030   ff 00 00 00 04 01 00 30 ea e4 31 a0 97 f5 2a 73
0040   67 06 70 34 53 e5 ce 04 01 00 84 99 2f ac b4 8d
0050   0e 92 90 19 f1 cf ca cc 9c 19 00 00 00 00 80 4f
0060   32 41 ff ff ff ff ff ff ff ff 01 00 00 00 1b 00
0070   00 00 6f 70 63 2e 74 63 70 3a 2f 2f 31 39 32 2e
0080   31 36 38 2e 35 2e 34 37 3a 34 38 34 30 16 00 00
0090   00 75 72 6e 3a 61 64 66 77 65 62 3a 6f 70 63 75
00a0   61 73 65 72 76 65 72 11 00 00 00 68 74 74 70 3a
00b0   2f 2f 61 64 66 77 65 62 2e 63 6f 6d 03 02 00 00
00c0   00 65 6e 0d 00 00 00 4f 50 43 20 55 41 20 53 65
00d0   72 76 65 72 00 00 00 00 ff ff ff ff ff ff ff ff
00e0   ff ff ff ff ff ff ff ff 01 00 00 00 2f 00 00 00
00f0   68 74 74 70 3a 2f 2f 6f 70 63 66 6f 75 6e 64 61
0100   74 69 6f 6e 2e 6f 72 67 2f 55 41 2f 53 65 63 75
0110   72 69 74 79 50 6f 6c 69 63 79 23 4e 6f 6e 65 02
0120   00 00 00 17 00 00 00 61 64 66 77 65 62 2d 61 6e
0130   6f 6e 79 6d 6f 75 73 2d 70 6f 6c 69 63 79 00 00
0140   00 00 ff ff ff ff ff ff ff ff ff ff ff ff 16 00
0150   00 00 61 64 66 77 65 62 2d 75 73 65 72 6e 61 6d
0160   65 2d 70 6f 6c 69 63 79 01 00 00 00 ff ff ff ff
0170   ff ff ff ff ff ff ff ff 41 00 00 00 68 74 74 70
0180   3a 2f 2f 6f 70 63 66 6f 75 6e 64 61 74 69 6f 6e
0190   2e 6f 72 67 2f 55 41 2d 50 72 6f 66 69 6c 65 2f
01a0   54 72 61 6e 73 70 6f 72 74 2f 75 61 74 63 70 2d
01b0   75 61 73 63 2d 75 61 62 69 6e 61 72 79 00 ff ff
01c0   ff ff ff ff ff ff ff ff ff ff 00 00 00 00
```

[image]

magiconair, 2025-05-01:
This looks suspicious. Looks like the clock on your server isn't configured correctly

```
Jan 1, 1970 03:08:19.000000000 RTZ 2 (зима)
```

HeczZots, 2025-05-01:
> This looks suspicious. Looks like the clock on your server isn't configured correctly
> 
> ```
> Jan 1, 1970 03:08:19.000000000 RTZ 2 (зима)
> ```
I have changed options to this
	opts := []opcua.Option{
		opcua.Lifetime(10 * time.Second),
		opcua.DialTimeout(30 * time.Second),
		opcua.AutoReconnect(true),
		opcua.ReconnectInterval(30 * time.Second),
		opcua.SessionTimeout(3600000 * time.Millisecond),
	}
now i can be connected and read for a while 
after 5 retries connection lost 
can i ignore server time by some of options or hacks ?
this server is a KNX converter to OPC UA which cant sync time from NTP server by no reason

[image]

magiconair, 2025-05-01:
It might not be able to sync the time because the clock is too far off. Try setting it with ntpdate first. You can also use `1 * time.Hour` as `SessionTimeout`.

HeczZots, 2025-05-01:
> It might not be able to sync the time because the clock is too far off. Try setting it with `ntpdate` or manually first. You can also use `1 * time.Hour` as `SessionTimeout`.

the problem is what i cant configure server time on controller

Also i detected what if ill connect via Ua Expert after i can connect via your golang package, i dont know why

but i can send you whireshark dumps

magiconair, 2025-05-01:
Sorry, I missed that in your first connect setup. I would just omit that value and then it defaults to 20 minutes. 

```
opcua.SessionTimeout(120000), // 120,000 * time.Nanosecond == 120 micro seconds
```

Does the `examples/read/read.go` work with `go run examples/read/read.go -node i=2258 -endpoint opc.tcp://...` ?

HeczZots, 2025-05-01:
> Sorry, I missed that in your first connect setup. I would just omit that value and then it defaults to 20 minutes.
> 
> ```
> opcua.SessionTimeout(120000), // 120,000 * time.Nanosecond == 120 micro seconds
> ```
> 
> Does the `examples/read/read.go` work with `go run examples/read/read.go -node i=2258 -endpoint opc.tcp://...` ?
> 
> Try with the minimal set of options and use the defaults. Only change them if necessary.

PS C:\Users\Trifonov.r\opcua> go run examples/read/read.go -node i=2258 -endpoint opc.tcp://192.168.5.47
The operation timed out. StatusBadTimeout (0x800A0000)
exit status 1

the same error

i cant find in code where is yours package validate timestamp from the server

HeczZots, 2025-05-01:
>
`> PS C:\Users\Trifonov.r\opcua> go run examples/read/read.go -node i=2258 -endpoint opc.tcp://192.168.5.47 The operation timed out. StatusBadTimeout (0x800A0000) exit status 1`
> 
> the same error
> 
> i cant find in code where is yours package validate timestamp from the server

HeczZots, 2025-05-01:
if i not specify any options i stack here

[image]

magiconair, 2025-05-02:
I'd need to see the full tcpdump of the conversation but I think I know what is going on. The timestamp of the `SecurityToken` in the `OpenSecureChannelResponse` is way in the past (1970). This triggers the immediate expiration of the channel since the token is too old. The timer expires immediately (https://github.com/gopcua/opcua/blob/main/uasc/secure_channel.go#L866). You should see an error message when you run `go run examples/read/read.go -debug ...`

However, I'm still trying to figure out how this actually leads to the `StatusBadTimeout` which should happen here (https://github.com/gopcua/opcua/blob/main/uasc/secure_channel.go#L944).

HeczZots, 2025-05-03:
```
trifonov@DESKTOP-GIA0VUQ:~/opcua$ go run examples/read/read.go -debug -node i=2258 -endpoint opc.tcp://192.168.5.47:4840
debug:  conn.go:70 uacp: connecting to opc.tcp://192.168.5.47:4840
debug:  conn.go:94 uacp 1: start HEL/ACK handshake
debug:  conn.go:429 uacp 1: sent HELF with 59 bytes
debug:  conn.go:388 uacp 1: recv ACKF with 28 bytes
debug:  conn.go:271 uacp 1: recv &uacp.Acknowledge{Version:0x0, ReceiveBufSize:0x4e20, SendBufSize:0x4e20, MaxMessageSize:0x14000, MaxChunkCount:0x14000}
debug:  secure_channel.go:576 sc.open
debug:  secure_channel_instance.go:89 got sequence number 1
debug:  secure_channel.go:1050 uasc 1/1: send *ua.OpenSecureChannelRequest with 132 bytes
debug:  conn.go:388 uacp 1: recv OPNF with 135 bytes
debug:  secure_channel.go:350 uasc 1/1: recv OPNF with 135 bytes
debug:  secure_channel.go:289 uasc 1/1: recv *ua.OpenSecureChannelResponse
debug:  secure_channel.go:304 uasc 1/1: sending *ua.OpenSecureChannelResponse to handler
debug:  secure_channel.go:660 OpenSecureChannelResponse handler
debug:  secure_channel.go:670 sc.handleOpenSecureChannelResponse
debug:  secure_channel.go:698 uasc 1: received security token. channelID=18 tokenID=18 createdAt=1970-01-01T00:02:43Z lifetime=10m0s
debug:  secure_channel_instance.go:89 got sequence number 2
debug:  secure_channel.go:836 uasc 1: security token is refreshed at 2025-05-03T09:15:06Z (7m30s). channelID=18 tokenID=18
debug:  secure_channel.go:1050 uasc 1/2: send *ua.CreateSessionRequest with 266 bytes
debug:  secure_channel.go:868 uasc 1: security token expires at 1970-01-01T00:15:13Z. channelID=18 tokenID=18
debug:  conn.go:388 uacp 1: recv MSGF with 462 bytes
debug:  secure_channel.go:287 uasc 1/0: err: opcua: sechan: unable to find instance for SecureChannelID=18
debug:  secure_channel.go:295 uasc 1/0: no handler for <nil>
debug:  secure_channel.go:1180 uasc 1: Close()
debug:  secure_channel_instance.go:89 got sequence number 3
debug:  secure_channel.go:1050 uasc 1/3: send *ua.CloseSecureChannelRequest with 57 bytes
debug:  conn.go:221 uacp 1: close
debug:  secure_channel.go:329 uasc 1: readChunk EOF
The operation timed out. StatusBadTimeout (0x800A0000)
exit status 1
```
i have opload pcapng
https://limewire.com/?referrer=pq7i8xx7p2

closed 2026-02-08 by github-actions[bot]