gopcua/opcua #801: Refactor server.MapNamespace

magiconair, 2025-05-14:
A couple of observations on `MapNamespace` while I was working on #800 

* make `Mu` and `Data` private
* guard access to `Data` through a method
* pass in `ExternalNotification` through the constructor method

danomagnum, 2025-05-15:
Making the Mu private makes it more difficult to actually use the map in a convenient way - you can't just lock it then iterate or whatever.  So I would propose something like this where I added iterators (named the same way as the iterators in the map package).  I also started playing around with making it generic for even more usability.

https://github.com/gopcua/opcua/compare/main...MapNamespaceRefactor

I haven't had a chance to test this or anything yet.

magiconair, 2025-05-15:
1. That looks like a reasonable approach. 

You could also make `SetValue` accept varargs, e.g.

```go
type KV struct {
	Key string
	Val any
}

func (ns *MapNamespace) SetValue(kvs ...KV) {
	ns.Lock()
	var nodes []*ua.NodeID
	for _, kv := range kvs {
		ns.Data[kv.Key] = kv.Val
		nodes = append(nodes, ua.NewStringNodeID(ns.id, kv.Key))
	}
	ns.Unlock()
	for _, node := range nodes {
		ns.srv.ChangeNotification(node)
	}
}
```

2. `SetValue` should take a `context` since `ChangeNotification` writes to a channel. Same goes for the write to the external notif channel. 

3. I'd inject the external notif channel since that lets the caller decide whether the channel is buffered or not. Also, make it write only.

```go
type MapNamespace struct {
	mu   sync.Mutex
	data map[string]any

	srv   *Server
	name  string
	id    uint16
	notif chan<- string
}

func NewMapNamespace(srv *Server, name string, notif chan<- string) *MapNamespace {
	ns := &MapNamespace{
		data:  make(map[string]any),
		srv:   srv,
		name:  name,
		notif: notif,
	}
	srv.AddNamespace(ns)
	return ns
}
```